rusefi/rusefi #7356: burn failed in bench test

ggurov, 2025-01-25:
**Describe the bug**
specifying value less than 5ish ms in OFF time in bench test causes burn error, and real funky aftereffects

**Hardware**
 - Board: any (mine was h7, same reproduced on f4)
 - Engine info bench

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Bench Test
2. Click on 'Off TIme"
3. Enter 2
4. Click burn
5. Observe error
6. POWER CYCLE ECU
7. Observe controller send 500ms as its internal Off time: 
8. Click "send current TS settings" this will send and burn 2 as offtime
9. Observe Error Again
10. After reload, accept what is in ECU
11. Go back to "Bench Test"
12. Observe "OFF TIME" read 500ms
13. Enter 50 ms into Off time
14. Click Burn - No error
15. Enter 5 ms  into OFF TIME -> burn -> No error
16. Enter 4 ms  into OFF TIME -> burn -> ERROR
17. Enter 5 ms into OFF TIME AGAIN -> burn -> ERROR
18. Enter 50 MS into off time -> BURN -> NO error

[image]**Additional context**Add any other context about the problem here.

Immrpablo, 2025-01-25:
from case #7355 - 

https://github.com/rusefi/rusefi/issues/7355

rusefillc, 2025-01-25:
@FDSoftware points at the explanation

[image]