rusefi/rusefi #7369: 36-1 RPM problem, no trigger error and SyncLatestRatio seems fine

ElDominio, 2025-01-30:
**Describe the bug**
Trying to run a 36-1 Ford engine gives me RPM dropouts, but triggerSyncLatest ratio is within expected bounds (0.7-1.3; 1.5-2.5) and I get no trigger errors. RPM shows spikes to 0 and car runs poorly. Hardware uses 9924 for VR.

**Hardware**
 - Board: Custom F407
 - Engine info Protege Turbo

**To Reproduce**
Steps to reproduce the behavior:
1. Set up trigger as 36-1
2. RPM spikes to zero

**Expected behavior**
No spikes

**Screenshots**

[image]
 

**Additional context**

[tunefor36and1.zip](https://github.com/user-attachments/files/18610056/tunefor36and1.zip)

[logfor36and1.zip](https://github.com/user-attachments/files/18610058/logfor36and1.zip)

rusefillc, 2025-01-31:
Your crank is happy, your cam is affecting you negatively

Note happy trigger error counter, note very unhappy resync counter @ElDominio 

[image]

ElDominio, 2025-01-31:
Hi, I'm Norman and welcome to Jackass

closed 2025-01-31 by ElDominio

ElDominio, 2025-01-31:
Can't we count cam resync errors, and just stop listening to cams after X amount of errors? Or maybe turn on the check engine light? (lol)

mck1117, 2025-01-31:
> Can't we count cam resync errors, and just stop listening to cams after X amount of errors? Or maybe turn on the check engine light? (lol)

Trying to magically "fix" your way out of a completely wrong configuration is a mistake.

ElDominio, 2025-01-31:
> > Can't we count cam resync errors, and just stop listening to cams after X amount of errors? Or maybe turn on the check engine light? (lol)
> 
> Trying to magically "fix" your way out of a completely wrong configuration is a mistake.

Which is why I said "light up the CEL bulb"

And are fallbacks not a thing? We just give up on handling user errors? Is that why I had to complain for years before the thing with duty cycles magically got done? User errors can be handled without having the car run like shit due to them IMO