rusefi/rusefi #7371: Target AFR Blend should be called Target AFR Adder

ElDominio, 2025-01-31:
**Describe the bug**
From what I can see, AFR Blend ADDS its value to the main AFR table, meaning having the table filled with 0.1 and your bias to 100% means your target AFR would be (value from main AFR Target + 0.1)

This should be reflected in title, if not it would not operate like this.

ElDominio, 2025-01-31:
[image]

rusefillc, 2025-02-01:
```
	// Add any blends if configured
	for (size_t i = 0; i < efi::size(config->targetAfrBlends); i++) {
		auto result = calculateBlend(config->targetAfrBlends[i], rpm, load);

		engine->outputChannels.targetAfrBlendParameter[i] = result.BlendParameter;
		engine->outputChannels.targetAfrBlendBias[i] = result.Bias;
		engine->outputChannels.targetAfrBlendOutput[i] = result.Value;

		target += result.Value;
	}
```

@ElDominio yes this blend is adding

@ElDominio please help by specifically pointing at exactly all the text you want to be replaced with exactly what text?

rusefillc, 2025-02-01:
@ElDominio does https://github.com/rusefi/rusefi/commit/98abdf190202f1aebba153f2dbb7a425b744823d cover it all?

ElDominio, 2025-02-01:
Good!

closed 2025-02-01 by ElDominio