rusefi/rusefi #7375: GDI: HPFP: activatio time is always fixed

dron0gus, 2025-01-31:
On https://github.com/rusefi/rusefi/blob/e2c4af7de82f6c3c3a42d872a4350fe078d665d9/firmware/controllers/engine_cycle/high_pressure_fuel_pump.cpp#L226 we calculate angle of next activation of HPFP using previously calculated `angle_requested`
But on `pinTurnOn` https://github.com/rusefi/rusefi/blob/e2c4af7de82f6c3c3a42d872a4350fe078d665d9/firmware/controllers/engine_cycle/high_pressure_fuel_pump.cpp#L193C22-L193C31 calback we ignore this value and use `engineConfiguration->hpfpActivationAngle`.
This seems incorrect for me.
After adding some debug https://github.com/rusefi/rusefi/pull/7374 I can confirm that HPFP activation time is constant (in angle units) independently of calculated `m_requested_pump` angle.
See delta between red and white lines on lowest graph.
[image]

dron0gus, 2025-01-31:
@racer-coder can you take a look?

dron0gus, 2025-01-31:
Ok. It seems my understanding of how this pump works is incorrect. It does not stop fuel delivery until top on lobe. Not until voltage is removed...

racer-coder, 2025-01-31:
Yeah, I don't know the physical reason for it, but yes once "latched" it keeps pressurizing until the top of the lobe.  Often there's a physical safety, like a valve that opens at 250 bar if normal operating pressure is 200 bar.  That's necessary because on shutdown, residual heat in the engine can heat the fuel thus increasing it's pressure.

I try to describe some of it here though I'm sure there are better descriptions floating around: https://rusefi.com/forum/viewtopic.php?f=5&t=2192
You can also check out some of the other aftermarket ECUs that support GDI, like Motec M1, Link G4+ Force GDI, Link G5 Voodoo, or now ECUMaster Pro-8 (at least the beta software) to see how they configure the pump solenoid operation.

racer-coder, 2025-01-31:
Hmm I wonder if this is why I've had trouble getting high fuel flow out of my car (using Link G4+ Force GDI).  It's angle based, but a certain angle means different fuel flow depending where you are on the lobe.  I'm assuming something about the latching mechanism involves a difference in pressure between the pressurizing piston and the low pressure rail (or maybe the high pressure rail), and the amount of time that takes depends on how much the piston moves i.e. how steep the lobe is at that point.

It wouldn't surprise me if we need a better strategy here, some combination of dead time in ms and movement of pump piston in % or ..... Is there a specific problem you're trying to diagnose?

rusefillc, 2025-01-31:
@racer-coder so far we have a few 4 cylinder cars (kia, passat, malibu) where we were able to locate a "good enough" magic angle and that was _all_ we needed to get relatively reasonable actual target

couple of days ago I failed to get nice actual pressure (by myelf, without extra sets of eyes or brains, by just eye-balling OEM oscilloscope vs rusEFI pump control oscilloscope) 

so I've raised the question "can we a) understand all this better b) improve data logging c) any other ideas towards a repetitive procedure of tuning and validating HPFP control"

I was touching 2017 GMC Sierra 5.3

[2025-01-28_13.52.55.zip](https://github.com/user-attachments/files/18624848/2025-01-28_13.52.55.zip)

my tune is in the bottom of gm_sbc.cpp file

rusefillc, 2025-01-31:
PS: @racer-coder https://rusefi.com/forum/viewtopic.php?p=50371#p50371

racer-coder, 2025-01-31:
Looking at the picture in that thread: https://rusefi.com/forum/download/file.php?id=11871
I would caution you to ignore everything before the cursor.  You can see the engine sync changed dramatically in the second or third cycle, so the fact that it reached 40 bar then is not very helpful.  But, you can see after the cursor it gained a little.  So you're close.

Knowing nothing about this, I would suggest logging (and graphing) these:

entry = m_requested_pump, "GDI: HPFP duration", float,  "%.3f"
entry = fuel_requested_percent, "GDI: requested_percent", float,  "%.3f"
entry = fuel_requested_percent_pi, "GDI: percent_pi", float,  "%.3f"

I could write a suggested plan of attack though I bet I have in the past, and it'd be very hackish at best, but it'd be a starting point.

One thing I will point out (based on the starting thought of this thread) is that if you are too EARLY in requesting the HPFP, you get nothing.  The solenoid will be turned off before the pump actually gets actuated.

But how you proceed from here depends on whether you want to just get THIS car running, or whether you want to a plan of attack that you then use to START OVER, applying it systematically to this car to make sure the plan of attack is a good one.

rusefillc, 2025-01-31:
@racer-coder any chance you can locate existing write-up? sounds like it's time to move it to some move visible place maybe on the wiki

attached logs have these fields

[image]

right now the goal is to develop a "base tune" specifically for three very specific engines: 2018ish GM 2.0, 5.3 and 6.2 engines

racer-coder, 2025-02-01:
hmm can't find it.  Might have to write it from scratch.

BTW going back to this log: https://rusefi.com/forum/download/file.php?id=11871
those injector times are awfully long, unless I'm misreading it.  133ms during crank?  This might simply be an issue of asking too much from the HPFP.  What does the oil smell like?  You might have washed and scored the cylinder walls.

rusefillc, 2025-02-01:
>  133ms during crank? This might simply be an issue of asking too much from the HPFP.

i believe 133ms was not actually happening because GDI driver has timeouts for injectors and pump https://github.com/rusefi/rusefi/issues/7370

racer-coder, 2025-02-01:
fair enough, but at 745RPM idle I'm at 0.67% injector duty cycle at 50 bar pressure.  That's 0.5ms by my calculations.  Yeah it should be longer if your rail isn't at 50bar but some random web calculator I found said about 3x longer at 6bar.  Unfortunately I don't know what % HPFP is needed to feed that.  

I don't have a log of start though.  I would try to address that (maybe even disable injection?) first and see if that helps the HPFP reach and stay at its target.

I kinda believe (but don't know for a fact) that you should be able to start and idle a direct injection engine without the HPFP at working pressure.  It should let the pressure from the lift pump bleed through it.

mck1117, 2025-02-01:
I drove my bone stock 2013 Focus ST once with a failed HPFP (failed while I was driving), and it drove completely normally on the LP pump except that it had maybe 50hp due to only 70psi of fuel pressure.

rusefillc, 2025-02-02:
@racer-coder @dron0gus https://rusefi.com/forum/viewtopic.php?p=50378#p50378 has three sets of logs

racer-coder, 2025-02-03:
Attempt at GDI/HPFP documentation.  Please comment on the forum thread.
https://rusefi.com/forum/viewtopic.php?f=16&t=2886