rusefi/rusefi #7376: knock is broken

rusefillc, 2025-02-01:
unhooked Proteus on bench, September build - note at least some random numbers

[image]

same proteus same bench, current build - nothing :(

[image]

rusefillc, 2025-02-01:
@dron0gus we had touched this area in October around https://github.com/rusefi/rusefi/commit/e06cd82c2667771522b3c46e72eb702c7f998cd4

ElDominio, 2025-02-02:
I had seen this, but I thought it was my hardware

rusefillc, 2025-02-02:
@dron0gus still zeros?

[image]

rusefillc, 2025-02-02:
[image]

closed 2025-02-02 by rusefillc