rusefi/rusefi #7380: isFanOn and isFan2On are gone, what replaces them?

ElDominio, 2025-02-02:
**Describe the bug**
I found a bunch of stuff used for logging that under "View" give workable values in TunerStudio, but in Lua they return "2147483647.7483647"  which is definitely not useable. Maybe find why Lua returns crazy values and make it the same value as TS?

**Hardware**
 - Board: custom f427
 - Engine info 4b11

**To Reproduce**
Steps to reproduce the behavior:
1. Used getOutput("isFanOn")
2. see it's broken
3. cry about not having fan state

**Expected behavior**
I can grab required fan state

**Screenshots**

[image]

rusefillc, 2025-02-14:
@ElDominio I am overwhelmed with a million of other cuts right now :(

rusefillc, 2025-02-26:
TL,DR: this is half broken half somewhat functional and I was hoping that sufficient bread crumps were left for self-service

@ElDominio Lua in general is documented at https://github.com/rusefi/rusefi/wiki/Lua-Scripting

Ctrl-F points one at https://github.com/rusefi/rusefi/wiki/Lua-Scripting#user-settings which points us at 

leveraging same Ctrl-F at https://github.com/rusefi/rusefi/blob/master/firmware/controllers/lua/generated/output_lookup_generated.cpp points as at

``
// cranking
#if FULL_SD_LOGS
		case 182391090:
			return engine->module<FanControl1>()->cranking;
#endif
// notRunning
#if FULL_SD_LOGS
		case -496961865:
			return engine->module<FanControl1>()->notRunning;
#endif
// disabledWhileEngineStopped
#if FULL_SD_LOGS
		case 142778571:
			return engine->module<FanControl1>()->disabledWhileEngineStopped;
#endif
// brokenClt
#if FULL_SD_LOGS
		case 545954889:
			return engine->module<FanControl1>()->brokenClt;
#endif
// enabledForAc
#if FULL_SD_LOGS
		case 291043035:
			return engine->module<FanControl1>()->enabledForAc;
#endif
// hot
#if FULL_SD_LOGS
		case 193494032:
			return engine->module<FanControl1>()->hot;
#endif
// cold
#if FULL_SD_LOGS
		case 2090155911:
			return engine->module<FanControl1>()->cold;
#endif
// disabledBySpeed
#if FULL_SD_LOGS
		case 2105103113:
			return engine->module<FanControl1>()->disabledBySpeed;
#endif
// m_state
#if FULL_SD_LOGS
		case -1337468622:
			return engine->module<FanControl1>()->m_state;
#endif
// radiatorFanStatus
#if FULL_SD_LOGS
		case -1987291756:
			return engine->module<FanControl1>()->radiatorFanStatus;
#endif

``

m_state is the horrible but answer for fan1 right now. Let me improve things a bit hopefully very soon.

rusefillc, 2025-02-26:
https://github.com/rusefi/rusefi/wiki/Lua-Scripting#user-settings now links at https://github.com/rusefi/rusefi/blob/master/firmware/controllers/lua/generated/output_lookup_generated.cpp which has both fan1 and fan2 data points

ElDominio, 2025-02-27:
How can I call m_state from lua? there are two FanControl instances, 1 and 2, how can I call this from lua?

ElDominio, 2025-02-27:
According to https://github.com/rusefi/rusefi/blob/master/firmware/controllers/lua/generated/output_lookup_generated.cpp , 
i need to call "fan1m_state"

such a call returns this

[image]

ElDominio, 2025-04-12:
Why is this closed?

I have a 4chan revi with a PNP harness which I cannot update due to fears of losing fan
@rusefillc

rusefillc, 2025-04-12:
ticket was closed because some progress was in fact made - m_state is available on F7 builds

We are out of resources on F4. I can make rearrangement to have fan available for Lua outputs but big picture is this: F4 is on the way out. Meaning that for instance silver boards with F4 is a mistake, meaning that fabricating parallela with F4 is not a very wise decision. 

Another angle of closing tickets: @ElDominio on average you do not provide me with test feedback in a timely manner, on average once something is addressed my impression is that I would never hear about it - including not hearing that yes, you confirm fix. That definitely contributes to me just closing when I assume that sufficient chance of progress exists.

rusefillc, 2025-04-12:
Just reshuffled what is being cut and what is left on F4 builds, closing

https://github.com/rusefi/rusefi/commit/349cd3d68f80e36866516eaef7bb25070c0b3555

closed 2025-04-12 by rusefillc

ElDominio, 2025-04-12:
> ticket was closed because some progress was in fact made - m_state is available on F7 builds
>
> We are out of resources on F4. I can make rearrangement to have fan available for Lua outputs but big [...]

What about all the uAEFI out there on F407? I think there is a gigantic user base on F407 that will have issues due to space constraints they are unaware of.

A question about F4, is F427 not the same amount of memory as F7?

rusefillc, 2025-04-12:
@ElDominio https://github.com/rusefi/rusefi_documentation/commit/6bd1d0b5ba3e829fac86c5bafa5e671dd275a9a1

ElDominio, 2025-04-13:
By the magic of sitting down and reading documentation, I have successfully compiled my own build of F4 locally with a bunch of stuff removed (mainly the image of wideband firmware) and actually cut BOTH fan state and AC state from being callable by setting them both to only be accessible when FULL_SD_LOGS is true, and then hijacked getDigital to also provide me fan and compressor states directly without depending on getOutput.

[image]

I believe this will allow me to have both the free RAM of not having these variables running at all times, while allowing me the needed Lua access. Is there interest for me to at least push the changes to getDigital? I have massacred other functions which i'm sure should not be massacred on main branch so I'd have to figure out how to do this specifically

rusefillc, 2025-04-13:
@ElDominio I understand where you are coming from but ``getDigital`` is not the right direction. If anything better direction would be https://github.com/rusefi/rusefi/commit/ec776b67cfa9964f860aa822f859920eb9cbd2bc

ElDominio, 2025-04-15:
> > [@ElDominio](https://github.com/ElDominio) I understand where you are coming from but `getDigital` is not the right direction. If anything better direction would be [ec776b6](https://github.com/rusefi/rusefi/commit/ec776b67cfa9964f860aa822f859920eb9cbd2bc)
> 
> [image]
> 
> I have done this, i will test it in the coming days and report if it is useful at all, thanks for the recommendation!

getFan and getFan2 confirmed working! thanks again