rusefi/rusefi #7383: [Usability] veAnalyze should stop correcting when DFCO is Active

benassi124, 2025-02-03:
If someone is using veAnalyze and they have DFCO active, upon decel veAnalyze will try to compensate for the fuel being shut by adding crazy amounts of fuel. veAnalyze allows you to lock cells so no corrections are made, the problem is a lot of people forget this step.

I am often frustrated when I have to correct VE tables that are messed up because someone forgot to lock out the "Decel" region of the VE Map in VE analyze and DFCO is active.

If possible we should make it so that when DFCO is active, veAnalyze is not making corrections.

joshImpact25, 2025-02-04:
normally this is done by setting the dfco to a specific map pressure setting,  pv   instructs you to set it   just below idle map  to prevent this  from happening.   this could be a setting in the autotune parameters  as it does change with large cams  or different displacements

joshImpact25, 2025-02-05:
{Useability} if it isn`t too much trouble adding a max map  value  would be nice as well as most wide open tuning is done manually by the tuner anyways.