rusefi/rusefi #7444: UAEFI121 INJ_8 not working

Artux0418, 2025-02-13:
**Describe the bug**
A clear and concise description of what the bug is.

**Hardware**
 - Board: UAEFI121
 - Engine info BMW m62

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Injector hardware
2. Click on Injection Output 8
3. Scroll down to INJ_8
4. See error 

**Expected behavior**
INJ 8 pin not working

**Screenshots**

[image]

**Additional context**
Add any other context about the problem here.

rusefillc, 2025-02-13:
@Artux0418 thank you for your defect report!

please either reset firmware and calibrations, or manually un-select CAN2 pins which are part of the problem here. firmware would not longer auto-select CAN2 pins.

Artux0418, 2025-02-14:
Unfortunately, the problem still exists. The error refers to foot 1. How can I turn it off? I can't find the setting in a tunerstudio.

[image]

rusefillc, 2025-02-14:
Please hit 'popular vehicles' honda obd

Artux0418, 2025-02-14:
I don't have hond obd. 

[image]

rusefillc, 2025-02-14:
@Artux0418 please click "vag i5"

Artux0418, 2025-02-14:
[image]

[image]

Unfortunately, he still writes this mistake. Am I doing something wrong? I have also tried firmware and config reset. from the console.

Artux0418, 2025-02-15:
[image]

I may not have touched and thought badly. I’m still learning a lot now because I’m specifically interested and glad I can help.
 However, B13 is still marked as a CAN port in uaefi121 confikurácios files.

Artux0418, 2025-02-16:
Hi again!

I got some help with the investigation and came to the conclusion that the INJ 7 and INJ 8 outputs are incorrectly labeled.
I will also illustrate this with pictures.

On the PCB, the INJ 7 and INJ 8 injectors are connected to the U2 chip – I verified this with a multimeter.

Currently, INJ 7 actually controls INJ 8 according to the pinout,
while INJ 8 does not work on the INJ 8 output.

`
  - pin: 50a
    class: outputs
    meta: MM100_MEGA_UAEFI_SPI2_SCK
    ts_name: ___ INJ_7
    function: Injector 7
    color: white
    type: inj

  - pin: 51a
    class: outputs
    meta: MM100_SPI2_SCK
    ts_name: ___ INJ_8
    function: Injector 8
    color: green
    type: inj`

However, Injector 7 should be MM100_MEGA_UAEFI_SPI2_CS,
and Injector 8 should be MM100_MEGA_UAEFI_SPI2_SCK—at least that's what I think.

I might not necessarily be right. I haven’t been able to test it because my computer is acting up, and I couldn't set up the test environment.

[image]

[image]

rusefillc, 2025-02-17:
replaced https://github.com/rusefi/rusefi/issues/7457

commit af402964 (2025-02-13): UAEFI121 INJ_8 not working fix #7444
only:uaefi121
closed 2025-02-17 by rusefillc