rusefi/rusefi #7482: let's remove "overrideCrankingIacSetting" "Override cranking IAC CLT multiplier" flexibility and always use additional table

rusefillc, 2025-02-21:
at the moment base cranking IAC is multiplied by either running correction or dedicated cranking correction

Question: is that crazy complexity shall we remove the choice?

ElDominio, 2025-02-21:
[image]

I always lock both "base duty" at 50% and never think about them again, and i always use both base duty tables. They should just use the base duty number itself (so i don't have to think about 50% base duty * multiplier) and just have the duty cycle directly on the table.

rusefillc, 2025-02-22:
https://github.com/rusefi/rusefi/issues/7485 to merge cranking base value and mult

closed 2025-02-23 by rusefillc