rusefi/rusefi #7487: smarter USB serial disconnect

rusefillc, 2025-02-22:
we have "reboot ECU" button in TS

problem: Windows or TS are not happy in terms of USB serial

question - can we make a low effort change to more cleanly disconnect from USB while scheduling ECU reset? is that even a thing on the USB layer/would have help Windows and/or TS?

FDSoftware, 2025-02-23:
chibios supports completely disconnecting the USB interface which (I guess) would also disconnect the CDC drivers

https://chibiforge.org/doc/19.1/full_rm/group___u_s_b.html#ga36f3e43880b82451f7c884e82602163f

at ST LL level we also have `USBD_LL_Stop` and `USBD_LL_DeInit`

https://github.com/STMicroelectronics/STM32CubeF1/blob/master/Projects/STM3210E_EVAL/Applications/USB_Device/MSC_Standalone/Src/usbd_conf.c#L248-L275

I think these last ones are already available within the rusefi code to use them?

https://rusefi.com/docs/html/group___u_s_b_d___c_o_r_e___exported___functions_prototype.html

also more context about this issue for other people who read it:
https://forum.chibios.org/viewtopic.php?t=1727

mck1117, 2025-02-23:
> chibios supports completely disconnecting the USB interface which (I guess) would also disconnect the CDC drivers

simply resetting the chip is the same, the host detects that the device has gone missing

the problem with TS is that you have to wait for it to timeout, there's simply no serial port protocol for "this serial port is about to literally completely disappear" because in the time of hardware serial ports that was impossible