rusefi/rusefi #7495: cal backup failed - Ordinal out of range 500 in nitrousRelayPin while 328 UINT16

ggurov, 2025-02-24:
**Describe the bug**
Using firmware update via rusefi_console, sometimes current cal fails to back up, aborting the update process.

**Hardware**
 - Board: uaefi
 - Engine info: na

**To Reproduce**
Steps to reproduce the behavior:
1. run rusefi_console
2. auto blt update
3. See error - and fail to update

**Expected behavior**
1. proceed to update firmware, even though there was a cal backup error. in case it has to be overwritten
2. not have the error.

**Screenshots**

[image]

**Additional context**

updating from a custom firmware back to bundle from build_server, but this has come up before within same versions.

`Console version 20250221
Windows 11 10.0
Bundle BundleInfo{branchName='development', target='uaefi'}
Suspending port scanning...
Port scanning is suspended.
Connecting to port `COM43`...
Performing action on port COM43...
Received a signature rusEFI master.2025.02.24.uaefiBIGFUEL.3948586501
Page size is 23304
Reading current calibrations...
Backing up current ini-file `..\rusefi_uaefi.ini`...
`rusefi_uaefi.ini` ini-file is backed up as `state/prev_calibrations.ini`
Backing up calibrations to files `state/prev_calibrations.zip` and `state/prev_calibrations.msq`...
Backing up current calibrations failed: java.lang.IllegalStateException: Ordinal out of range 500 in nitrousRelayPin while 328 UINT16
Resuming port scanning...
Port scanning is resumed.
Failed to back up current calibrations...
hint: error state is already in your clipboard, please use PASTE or Ctrl-V while reporting issues
`

ggurov, 2025-02-24:
alternative to the fix, a more descriptive error showing me the error of my ways would also work as a fix.

kifir23917, 2025-02-25:
Status: today we are not able to reproduce this issue.

@ggurov, in case if you reproduce this problem, could you please:
1. archive content of `console\logs` and `console\state` folders
2. reopen ticket and attach archived file

closed 2025-02-25 by kifir23917

rusefillc, 2025-03-25:
see https://github.com/rusefi/rusefi/issues/7652