rusefi/rusefi #7502: second units specifically for injector flow

rusefillc, 2025-02-25:
Harley industry standard for fuel flow is grams per second and every one using this will be familiar with that

not even gallons per... per... what's freedom unit for time?

rusefillc, 2025-02-25:
@joshImpact25 @dialedinmotorsports forgive me for dumb question what's the conversion rate we will apply for fuel?

rusefillc, 2025-02-25:
g/s = 0.012 * cc/min ? not sure since I have "990; // 12.6g/sec" and math does not add up

benassi124, 2025-02-25:
Yes. G/S = cc/min * .012 for quick math I will find full math tomorrow.

rusefillc, 2025-02-25:
@FDSoftware probably following

```
#if LAMBDA
	gauge2		= lambda1Gauge
#else
	gauge2		= afr1Gauge
#endif
```

pattern. in MS they use this for C vs F maybe, anyway see MS .ini as reference?

FDSoftware, 2025-03-01:
> not even gallons per... per... what's freedom unit for time?

wikipedia says "Pound per hour (lb/h)" is the unit on US

https://en.wikipedia.org/wiki/Pound_per_hour#:~:text=Pound%20per%20hour%20is%20a,more%20conventionally%20as%20lb%2Fh.

joshImpact25, 2025-03-01:
For the Harley stuff industry standard is grams per second

rusefillc, 2025-03-01:
@FDSoftware it's a mess

* we already have a drop-down menu for units - including the desired grams per second
* but it has pretty... unclear naming
* and yes, there is the THIRD set of units which is lb/hr

I believe that minimal improvement is https://github.com/rusefi/rusefi/commit/8f42e81662708017047c78b3936aa3ba75552ce8

with that minimal improvement the whole urgency goes down a bit. Next open question is how to approach all three units

FDSoftware, 2025-03-01:
> but it has pretty... unclear naming

Yep, I thought that option configured something else, but it's much clearer now with that commit.

reading this on  `status_loop.cpp`

```cpp
static void updateFuelResults() {
#if EFI_VEHICLE_SPEED
	engine->outputChannels.fuelFlowRate = engine->module<TripOdometer>()->getConsumptionGramPerSecond();
	engine->outputChannels.totalFuelConsumption = engine->module<TripOdometer>()->getConsumedGrams();
.....
```
and: `injector_model.cpp`

```cpp
constexpr float convertToGramsPerSecond(float ccPerMinute) {
	return ccPerMinute * (fuelDensity / 60.f);
}

float InjectorModelWithConfig::getBaseFlowRate() const {
	if (engineConfiguration->injectorFlowAsMassFlow) {
		return m_cfg->flow;
	} else {
		return convertToGramsPerSecond(m_cfg->flow);
	}
}
```

and searching a bit about injectors for sale:

[image]

[image]

[image]

[image]

[image]

one can conclude that:

* All injectors sold in cc/min also have units in lb/h and vice versa
* The only ones adopting different measurements are Harleys that use g/s (and they don't show you the other two units)
* we already support the latter case

So there is not much use in supporting lb/h from the *configuring an injector* side [since each vendor offers you cc/m as well so you don't have to look for any conversion].

but maybe it is useful for visualization?

In my case I don't think so but I have bias since I live in a country that uses the metric system.