rusefi/rusefi #7511: injector lag is a table

rusefillc, 2025-02-25:
transition from

[image]

to

[image]

rusefillc, 2025-02-26:
@FDSoftware let's use 'FuelPressureInjector' for now

FDSoftware, 2025-02-26:
some notes:

at: `firmware/controllers/algo/defaults/default_fuel.cpp`
`setBosch02880155868` It will break because 8 values ​​are set instead of the 2 that this table will have. How to proceed? Restrict the function to only use the default 2x2?

at: `unit_tests/tests/ignition_injection/test_fuel_map.cpp`
`TEST(misc, testFuelMap) `
This is setting up a custom deadtime table, with these changes I'm going to have to edit this test in addition to the obvious one at `test_injector_model.cpp` / `TEST(InjectorModel, Deadtime)`

at: `firmware/config/engines/mazda/mazda_miata_vvt.cpp` A custom deadtime table is also being set up, will I have to look up some information about the injector used and restrict the table to 2x2?

rusefillc, 2025-02-26:
can we set both lines of default calibration to same value, with Y axis say 300kpa and 600 kpa?

meaning regardless of pressure defaults would have same exact runtime values in miata and bosch?

closed 2025-02-27 by rusefillc