rusefi/rusefi #7515: TUNING DETECTED indicator and pause STFT if tuning is detected

rusefillc, 2025-02-26:
problem: STFT negatively affects tuning sessions of new users

proposal: automatically detect that tuning seems to be happening based on tables being changed within last XXX-TBD seconds?

if tuning is detected we light up indicator and pause STFT

rusefillc, 2025-02-26:
@ElDominio @dialedinmotorsports @mck1117 opinions? can you fill in the XXX blank?

ElDominio, 2025-02-26:
I like this idea, i wonder if it would help tuners or if it would make them angry

ggurov, 2025-02-26:
this should be toggleable to account for ve live autotune. live ve autotune looks at both EGO corr and current afr

ggurov, 2025-02-26:
> I like this idea, i wonder if it would help tuners or if it would make them angry

it should be toggleable, default to ON is fine, as long as this sort of forced intervention can be switched off

ElDominio, 2025-02-26:
> > I like this idea, i wonder if it would help tuners or if it would make them angry
> 
> it should be toggleable, default to ON is fine, as long as this sort of forced intervention can be switched off

I like this, because sometimes i leave STFT on while i tune and watch trim change in real time

rusefillc, 2025-02-26:
See https://github.com/rusefi/rusefi/issues/7513

kifir23917, 2025-02-26:
We need to add boolean TUNER DETECTED indicator on STFT dialog.

commit e7428eab (2025-02-28): Place `Tuning Detected` indicator on `Short term fuel trim` dialog (closes #7515)
closed 2025-02-28 by kifir23917