rusefi/rusefi #7517: GDI fueling seems strange: Pulsewidth goes from 4ms to 28ms with only slight changes in VE/DI pressure

ElDominio, 2025-02-26:
**Describe the bug**
GDi car seems to be requesting crazy injection time for less than a second, a few times when accelerating. I have no accel enrich (I haven't seen DI ever need this) and no fuel-on-walls. Specifically in the log, at 4.1s and 5.5s you can see some crazy injeciton pulses while VE has only changed by 2%. Where is the math wrong? Where am I wrong?

**Hardware**
 - Board: 4K GDI with PNP Board
 - Engine: Hyundai Gamma

**To Reproduce**
Steps to reproduce the behavior:
1. Copy tune from PB base
2. accel for a second
3. wathc injection time grow massively

**Expected behavior**
No crazy injection times with basically linear VE

**Screenshots**

[image]

**Additional context**
Tune and log: https://rusefi.com/online/view.php?log=1436

ElDominio, 2025-02-26:
This might be because PB base was made with injection mode being End of Injection, and PB base from default tunes has it set to Center of Injection; this is now pending further testing

mck1117, 2025-02-26:
@rusefillc y'all have been playing VERY fast and loose with tune layout these days, and this is the result:

[image]

rusefillc, 2025-02-26:
> [@rusefillc](https://github.com/rusefillc) y'all have been playing VERY fast and loose with tune layout these days

Obviously I appreciate your help. Any chance you could connect the dots between layout changes and what's going on here?

@ElDominio would be great if you could invest time and re-trace your steps. If you choose to just fix the tune based on @mck1117 help we might end up in same again.

ElDominio, 2025-02-26:
This is a fresh ECU, and incompatible with 4chan tune

I loaded the PB base tune under "controller", then started inporting changes manually from my previous PB

I would think this was caused by using the PB base button?

rusefillc, 2025-02-26:
Closing since @ElDominio does not have resources to re-trace his steps. Unfortunately I do not have resources to work on reproducing this issue, so until next time!

closed 2025-02-26 by rusefillc

rusefillc, 2025-02-26:
PS: not a GDI fueling problem anyways

ElDominio, 2025-02-26:
> Obviously I appreciate your help. Any chance you could connect the dots between layout changes and what's going on here?
>
> @ElDominio would be great if you could invest time and re-trace your steps. [...]

You say "retrace your steps"

Do you want me to grab the ecu from a desk and press the PB button and get the engine started again? What else could i do? I'm failing to understand what else you expect me to provide you with.

Afaik @mck1117 found the issue with something about the AFR table loading which I didn't even open or import or do anything with.

rusefillc, 2025-02-26:
@ElDominio yes I absolutely want to grab the ecu from a desk and reset whatever reset means and re-apply all your steps and definitely confirm on what step AFR turned crazy. And then write the instruction for me so that I can repeat based on your assertion that I have all the steps to get broken result quicky and on demand here in NJ.

I assume that starting the engine is not required unless we would be unable to reproduce on desk without engine running.

ElDominio, 2025-02-26:
> @ElDominio yes I absolutely want to grab the ecu from a desk and reset whatever reset means and re-apply all your steps and definitely confirm on what step AFR turned crazy. And then write the [...]

I now understand your ask much better. I will try it tomorrow