rusefi/rusefi #7522: migrate injector lag from 1x8 to 2x8

rusefillc, 2025-02-26:
Similar to https://github.com/rusefi/rusefi/issues/7476 we should have conversion for https://github.com/rusefi/rusefi/pull/7520 incompatible change

rusefillc, 2025-03-07:
@kifir23917 here we just copy curve values into both lines of 2x8 table, i.e. each of older values would be used twice - once in top and once in bottom line of the table

kifir23917, 2025-03-07:
Last week we had [a commit](https://github.com/rusefi/rusefi/commit/cb82c85481d3e4f66b11f6b046be305bfb77e7cb#diff-569a1c62ada1e8d1ccc5b6d7295cfba191557b0b0ad8262d44bdee511a31f743R496) that renames `batLagCorr` constant to `batLagCorrTable` and our migration does not understand that these fields should be migrated. Are we going to enhance our migration procedure to handle such case?

commit 8be41c75 (2025-03-03): migrate injector lag is now a table fix #7522
only:uaefi
commit 040ea1d6 (2025-03-18): implement migration of `injector_battLagCorrBins` and `injector_battLagCorr` ini fields to `injector_battLagCorrBattBins` and `injector_battLagCorrTable` ini fields (closes #7522)
commit 9cf6b43c (2025-03-24): implement migration of `injector_battLagCorrBins` and `injector_battLagCorr` ini fields to `injector_battLagCorrBattBins` and `injector_battLagCorrTable` ini fields (closes #7522)

(cherry picked from commit 040ea1d6bc577b723b6bee38d67a34f6ed58b1df)
closed 2025-03-18 by rusefillc