rusefi/rusefi #7547: Scalable Tachometer output

ggurov, 2025-03-01:
**Is your feature request related to a problem? Please describe.**
Currently tachometer output is only full speed or half speed, this presents a problem for inaccurate tachometers or conversions from incompatible cylinder counts 4 -> 6, 6 -> 8, etc.

**Would we need new configuration fields?**
yes, two, variable tachometer (yes/no), and scaling percentage 

**Describe the solution you'd like**

[image]

**Describe alternatives you've considered**
Dealing with inaccurate period-tachometer, installing a digital dash, or just simply DGAF

Zazins, 2025-03-03:
This could be very good option. Or pulse by rev could be used in decimal values not only fixed value. As BMW using fixed values like 6 or 3 for analog clusters have high error in values.

rusefillc, 2025-03-04:
@ggurov please test

closed 2025-03-04 by rusefillc

rusefillc, 2025-04-23:
[image]

Zazins, 2025-04-23:
Hi! 

I have tested multiple clusters but as it seems it's not useful, however it works as it should.
On clusters I notice that even at low RPM it is able to correct the error but at high RPM error is back as it was in low RPM before. 
It's impossible to  achieve sweet middle spot. 
As a feature it can stay maybe for other kind clusters.
Thanks!

commit d51080f9 (2025-03-04): fractional tach pulse fix #7547
commit 038ba8a9 (2025-03-04): fractional tach pulse fix #7547