rusefi/rusefi #7556: '-softPrepend`

rusefillc, 2025-03-03:
in bash script we have 

```-prepend ${BOARD_DIR}/prepend.txt```

which currently required some empty files

we should do smarter and add a feature of '-softPrepend` which would allow for files not to exist

rusefillc, 2025-03-03:
end goal ``prepend_${SHORT_BOARD_NAME}.txt``

rusefillc, 2025-03-06:
@FDSoftware can you actually add two different proteus files with only lua script line?
```

! f4 does not like this #define LUA_SCRIPT_SIZE 32000
! TODO what we need is a way to define one LUA_SCRIPT_SIZE for F4 and different one for F7
! todo: maybe some integration between meta_info and prepend?
#define LUA_SCRIPT_SIZE 13500
```

FDSoftware, 2025-03-06:
> [@FDSoftware](https://github.com/FDSoftware) can you actually add two different proteus files with only lua script line?
 proteus H7 also need the small lua script size, i will add another prepend for that target

FDSoftware, 2025-03-06:
@rusefillc Can this be closed or do we have to do something extra?

closed 2025-03-06 by rusefillc