rusefi/rusefi #7563: make open loop idle a bit simper - remove multiplication

rusefillc, 2025-03-04:
```
engineConfiguration->manIdlePosition		// Base idle position (slider)
		* interpolate2d(clt, config->cltIdleCorrBins, config->cltIdleCorr)
```

let's replace slider + multiplier curve with just one curve for open loop value by CLT

FDSoftware, 2025-03-04:
test on:
`java_console/autotest/src/main/java/com/rusefi/proteus/ProteusAnalogTest.java`
is using command:

`idle_position`:
```cpp
void setManualIdleValvePosition(int positionPercent) {
	if (positionPercent < 1 || positionPercent > 99)
		return;
	efiPrintf("setting idle valve position %d", positionPercent);
	// todo: this is not great that we have to write into configuration here
	engineConfiguration->manIdlePosition = positionPercent;
}
```
However, the rest of the tests at HWCI use this newer method:

`idle_solenoid_freq` :
```cpp
#if EFI_ENGINE_CONTROL
static void setIdleSolenoidFrequency(int value) {
	engineConfiguration->idle.solenoidFrequency = value;
	incrementGlobalConfigurationVersion();
}
#endif // EFI_ENGINE_CONTROL
```

This seems a bit of a duplicate to me, although one command talks about position and another about PWM alone, at the same time the Proteus command generates a dependency on "manIdlePosition" which is the field to remove, as a first step do I edit the Proteus test to use the same function for HWCI as the rest of the tests and then remove the command?

rusefillc, 2025-03-04:
@FDSoftware incremental change _could_ be to set manIdlePosition to '1' and convert all multipliers to match? but that could not work due to limits etc, just a random idea which might be not so helpful

FDSoftware, 2025-03-04:
> [@FDSoftware](https://github.com/FDSoftware) solenoidFrequency probably has nothing to do with this at all?

yep, I answered myself just by reading `applyIACposition` func, I was a bit confused on how the valve position was set

> for setManualIdleValvePosition we would simply have to set complete array to whatever same value?

cool, i do it that way then

closed 2025-03-04 by rusefillc