rusefi/rusefi #7567: cranking base fuel and cranking coolant multiplier

rusefillc, 2025-03-04:
Is it trivial to combine cranking base fuel and cranking coolant multiplier similar to https://github.com/rusefi/rusefi/issues/7563 or would it get hard around "fuel source for cranking" option?

rusefillc, 2025-05-11:
@FDSoftware what's the status/plan here? Are we looking to combine these two?

[image]

FDSoftware, 2025-05-12:
> [@FDSoftware](https://github.com/FDSoftware) what's the status/plan here? Are we looking to combine these two?

They can be combined, but at the same time we have another curve that is used for engines with ethanol

`auto e85Mult = interpolate2d(clt, config->crankingFuelBins, config->crankingFuelCoefE100);`

```c++
engine->engineState.crankingFuel.coolantTemperatureCoefficient =
			interpolateClamped(
				0, e0Mult,
				85, e85Mult,
				flex
			);
```

TS description:

> Enable a second cranking table to use for E100 flex fuel, interpolating between the two based on flex fuel sensor.

[image]

If we want to combine them, we will have to update the description of how the e85 curve works, since it would become a direct multiplier and not interpolated with e0, is this okay?

ElDominio, 2025-06-13:
We need this to in effect only multiply Base Cranking Fuel, and it should really only be a single value than multiplies base cranking fuel

All coolant considerations are now under 

[image]

So we don't need ot think about Coolant temp anymore, only the change from normal petroleum fuel to E85-100