rusefi/rusefi #7575: max oil pressure protection

rusefillc, 2025-03-06:
we have min oil pressure protection

we shall add similar max oil pressure protection with a 4 element curve by RPM

default value 8bar across the line

rusefillc, 2025-03-06:
[image]

rusefillc, 2025-03-06:
@esaulenka step one could be adding a scalar setting, step two making a curve based on RPM

rusefillc, 2025-03-09:
@esaulenka you are way overthinking it

Please repeat low pressure protection, but make it high pressure 

Some oil pumps have control solenoids, if it fails pressure can get high

esaulenka, 2025-03-09:
Ok. But it looks like MAX logic is much simplier than MIN, since zero on startup is ok, but value above max is always error.

rusefillc, 2025-03-09:
@esaulenka add second similar one

rusefillc, 2025-03-09:
@esaulenka you've added fields into all relevant structures, please add those into UI by adjusting tunerstudio.template.ini file

while we are on that please implement phase two - let's migrate from one scalar setting to a curve by RPM. 4 elements in the curve would be sufficient.

esaulenka, 2025-03-09:
> add those into UI by adjusting tunerstudio.template.ini file
Ok, I need to learn it first.

> let's migrate from one scalar setting to a curve by RPM
you mean I need to replace this maxOilPressure with a curve ?

rusefillc, 2025-03-09:
@esaulenka yes, replace scalar with curve (two arrays)

closed 2025-03-24 by rusefillc