rusefi/rusefi #7595: Does HPFP Pump Compensation Factor do anything at all?

ElDominio, 2025-03-10:
**Describe the bug**
It is unclear what this table does. I have written values on it from 0-100 and nothing seems to change in the car (pump duty stays the same, injection time stays the same, pressure stays the same, afr stays the same). What does this "compensation" do?

**Hardware**
 - Board: Any GDI

**To Reproduce**
Steps to reproduce the behavior:
1. Open Controller
2. Click HPFP Pump Compensation Factor
3. Write values on table
4. look for changes

**Expected behavior**
any changes at all when values are entered

**Screenshots**

[image]

racer-coder, 2025-03-11:
See Step 7 here: https://rusefi.com/forum/viewtopic.php?f=16&t=2886
Basically you can rely on the PI controller to do all the work but you'll end up with bad performance on any transient condition.  Filling in the table with whatever the PI controller decided is the answer means that afterwards the PI controller has a lot less work to do.

rusefillc, 2025-03-11:
@racer-coder is this similar to what we can feedforward in ETB? @racer-coder would it be correct for me to replace "Compensation Factor" label with "Feedforward"?

racer-coder, 2025-03-12:
That would be fine, maybe less confusing.  It's not entirely correct but maybe only a pedant would care.

Feed Forward = a base map for a controller that the PI feedback works on top of

In this case (unlike ETB) you are asked to model the HPFP parameters - volume of pump, shape and position of lobe, etc.  So this is technically a compensation table on top of the base model.  One may ask why bother modeling the HPFP if this table exists and that is a valid point, but having the PI controller work in the domain of the model should provide a better result... in theory....

commit cca3d330 (2025-03-12): Does HPFP Pump Compensation Factor do anything at all? fix #7595
closed 2025-03-12 by rusefillc