rusefi/rusefi #7617: TCP proxy is not reliable

rusefillc, 2025-03-12:
have ECU on one device, run the following command to launch a proxy
```java -jar rusefi_console.jar local_proxy```

rusefillc, 2025-03-14:
in the meantime https://sourceforge.net/projects/ser2net/ or https://sourceforge.net/projects/com0com/files/com2tcp/

rusefillc, 2025-03-15:
@esaulenka see also https://gist.github.com/DraTeots/e0c669608466470baa6c#linux

esaulenka, 2025-03-19:
It's also not reliable enough with local_proxy replaced by ser2net utility.
Maybe you have some advices with it.

```
I 250319 200052.241 [main] Launcher - rusEFI UI console 20250312
I 250319 200052.260 [main] Launcher - Compiled Wed Mar 19 07:16:27 AMT 2025

...

I 250319 200054.313 [ECU Communication Executor1] BinaryProtocol - rusEFI says tune CRC32 0x2dcfdc48 768597064

I 250319 200054.313 [ECU Communication Executor1] BinaryProtocol - rusEFI says tune CRC16 0xdc48 -9144

I 250319 200054.315 [ECU Communication Executor1] BinaryProtocol - Got configuration from controller 47220 byte(s)
I 250319 200054.356 [AWT-EventQueue-0] BinaryProtocolServer - BinaryProtocolServer on 2390
I 250319 200054.357 [AWT-EventQueue-0] StatusConsumer - ServerSocket 2390 created. Feel free to point TS at IP Address 'localhost' port 2390
I 250319 200056.449 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: USB: got Query command
I 250319 200056.450 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: TS <- Query signature: rusEFI master.2025.03.10.proteus_f7.2433621198
I 250319 200056.450 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: TS <- Get CRC offset 0 count 47220 result 48DCCF2D
I 250319 200056.453 [ECU Communication Executor1] BinaryProtocol - Got livedata
I 250319 200056.454 [ECU Communication Executor1] ConnectionWatchdog - ConnectionWatchdog.reconnectTimer restarting: 1000
I 250319 200056.455 [ECU Commands Queue] BinaryProtocol - Sending [set date 2025-03-19T20:00:54.353982]
E 250319 200056.457 [[start] TCP reader1] ByteReader - TcpIoStream: End of connection java.net.SocketException: Socket closed
I 250319 200056.466 [ECU Communication Executor1] LinkManager - restart isPortAvailableAgain=false
I 250319 200056.552 [ECU text pull1] BinaryProtocol - Port shutdown: Stopping text pull
I 250319 200057.451 [ECU Communication Executor1] ConnectionWatchdog - ConnectionWatchdog.reconnectTimer restarting: 1000
I 250319 200057.453 [ECU Communication Executor1] LinkManager - restart isPortAvailableAgain=false

```

It can be reproduced in 100%: just after first `BinaryProtocol - Got livedata` I see watchdog message.

It's definitely not a version conflict, I have checked the same console that works locally on remote machine (even with the same connection through TCP).

rusefillc, 2025-03-19:
status update: tuner studio works via real network

esaulenka, 2025-03-20:
Increased watchdog timeout helped with this issue.

For some weird reason I can receive livedata only once in 2 seconds (about 2.2 sec to be preceise), and smaller watchdog values breaks all communication.

```
I 250320 170638.135 [ECU Communication Executor1] BinaryProtocol - Got livedata
I 250320 170640.290 [ECU Communication Executor1] BinaryProtocol - Got livedata
I 250320 170642.436 [ECU Communication Executor1] BinaryProtocol - Got livedata
I 250320 170644.485 [ECU Communication Executor1] BinaryProtocol - Got livedata
I 250320 170646.306 [ECU Communication Executor1] BinaryProtocol - Got livedata
I 250320 170648.482 [ECU Communication Executor1] BinaryProtocol - Got livedata
I 250320 170650.634 [ECU Communication Executor1] BinaryProtocol - Got livedata

```

rusefillc, 2025-03-20:
@dron0gus has mentioned TCP socket maybe https://github.com/rusefi/rusefi/issues/4202