rusefi/rusefi #7660: idle open loop curve to become a table

rusefillc, 2025-03-25:
we have a 16 element curve ``cltIdleCorrBins`` ``cltIdleCorr``, let's have 2x8 table

target RPM axis size 2
CLT axis size 8

[image]

dialedinmotorsports, 2025-03-25:
RPM axis of 2 is likely too small. This is a base feedforward table, at least 6x8 or 8x8 should be available

mck1117, 2025-03-26:
Modeled flow idle auto compensates for RPM.

dialedinmotorsports, 2025-03-26:
Even on modeled flow idle ECUs, specific Idle Target RPM has to be forward-fed a base idle flow. You can fully rely on the ECU doing the heavy lifting for you, but having 2-row RPM HEAVILY assumes that flow is linear with RPM... which in my experience with many engines, especially modified engines making 1000+HP this is not the case. It is proportional, yes, but not a linear interpolation.

Some engines will be fine with 2 RPM Points. Other engines will oscillate while in the mid-rpm area and heavily rely on PID "reacting" to control RPM.

rusefillc, 2025-03-26:
2 row is a RAM preservation technique, one day I need to check how we handle 1*X tables.

F4 units cannot afford many new tables, F7 units would have larger dimensions.

mck1117, 2025-03-26:
Modeled flow doesn't use flow as the open loop (and PID output), it uses airmass. Airmass required to idle the engine is very nearly constant and independent of RPM.

It then takes that airmass, multiplies by cylinder count and engine speed, then uses that to look up a valve position.

dialedinmotorsports, 2025-03-26:
I've yet to use an ECU, OEM or standalone, that bases the idle tuning strategy on air mass (i.e. g/cyl) instead of air mass flow (i.e. g/s) as the feed forward input on a table.

Always room for improvement though :)