rusefi/rusefi #7663: LUA getTsButtonCount() not supported in the simulator

esaulenka, 2025-03-27:
**Describe the bug**

The latest proteus_f7 simulator cannot load LUA that contains `getTsButtonCount()`

```
2025-03-27_11_10_19_644: EngineState: LUA ERROR loading script: [string "-- scriptname ats-mim-2018.lua..."]:413: attempt to call a nil value (global 'getTsButtonCount')
```

The same script works on a real HW.

For some reason this code is only under `EFI_PROD_CODE` define, maybe it would be better to add `EFI_SIMULATOR`
as well:
https://github.com/rusefi/rusefi/blob/b741aa1cda37cb2854be1aa5ab61da161f18dc6c/firmware/controllers/lua/lua_hooks.cpp#L759-L762

if I'm not mistaken, the second part is already in the simulator
https://github.com/rusefi/rusefi/blob/b741aa1cda37cb2854be1aa5ab61da161f18dc6c/firmware/controllers/bench_test.cpp#L336-L338

rusefillc, 2025-03-31:
fixed?

closed 2025-03-31 by rusefillc

esaulenka, 2025-04-01:
> fixed?

Unfortunately, no (cannot reopen this ticket).
And it looks like LUA in the simulator now completely broken.

If you open ` https://github.com/rusefi/rusefi/releases/download/2025-04-01/rusefi_bundle_proteus_f7.zip` and just press 'Write LUA' with the default script (that just print received CAN), you'll get
```
I 250401 122450.457 [ECU Communication Executor1] BinaryProtocol - Got livedata
I 250401 122450.457 [ECU Communication Executor1] LuaScriptPanel - Sending StringIniField{offset=5880, size=64000}
I 250401 122450.479 [ECU Commands Queue] BinaryProtocol - Sending [luareset]
I 250401 122452.176 [SimulatorExecHelper1] SimulatorExecHelper - terminate called after throwing an instance of 'std::logic_error'
I 250401 122452.176 [SimulatorExecHelper1] SimulatorExecHelper -   what():  Please set stoichRatioPrimary
E 250401 122452.176 [[start] TCP reader1] ByteReader - TcpIoStream: End of connection java.net.SocketException: Connection reset
I 250401 122452.231 [ECU text pull1] BinaryProtocol - Port shutdown: Stopping text pull
```

rusefillc, 2025-04-02:
@esaulenka ``what():  Please set stoichRatioPrimary`` is a totally unrelated defect. If you are able to reproduce please create a separate issue