rusefi/rusefi #7666: F7 MCU sends a 72ms pulse on all channels on boot, causing injectors to fire and coils to fire every power on

ggurov, 2025-03-30:
**Describe the bug**
F7 (and H7), when powered up, emit a 72ms on f7 (208ms on h7) on most ports
This causes injectors to inject 72ms of fuel on every power on event
F4 does not do this.

**Hardware**
 - Board: F7 (nucleo, custom), H7 custom
 - Engine info N/A

**To Reproduce**
Steps to reproduce the behavior:
1. Full erase F7 nucleo board
2. Flash base F767nucleo  firmware
3. Configure single ignition channel to any port (PD3)
4. Connect logic analyzer to the port
5. power down board
6. power up board,
7. see 72ms pulse on boot up

**Expected behavior**
Port to not initialize to a high state on boot

**Screenshots**

[image]

**Additional context**
F4 does not do this.

ggurov, 2025-03-30:
on h7, it sends a 208 ms pulse on most relevant channels

[image]

PC13 and PA8 are the channels that don't do this, for some reason, both H7 and F7

ggurov, 2025-03-30:
F7 has a 75ms pulse, same channels

[image]

F7 : only a 38ms if a reset event is sent after boot/power on

[image]

ggurov, 2025-03-30:
introducing a 4k7 pulldown on affected MCU pins fixes this, uncertain if there are firmware aspects that could be modified though

rusefillc, 2025-03-30:
this smells board.h

https://github.com/rusefi/rusefi/commit/d9adba2024ba477b7ea606e956ff92fea861cf28 https://github.com/rusefi/rusefi/commit/9cb08365a2a94cf7563113acf4cb2c83311c338d

looks like we shall follow the pattern on F7/H7?

rusefillc, 2025-03-30:
> PC13 and PA8 are the channels that don't do this, for some reason, both H7 and F7

  #define VAL_GPIOC_MODER             (PIN_MODE_INPUT(GPIOC_ARD_A1) |         \
                                       PIN_MODE_ALTERNATE(GPIOC_RMII_MDC) |   \
                                       PIN_MODE_INPUT(GPIOC_ZIO_A7) |         \
                                       PIN_MODE_INPUT(GPIOC_ARD_A2) |         \
                                       PIN_MODE_ALTERNATE(GPIOC_RMII_RXD0) |  \
                                       PIN_MODE_ALTERNATE(GPIOC_RMII_RXD1) |  \
                                       PIN_MODE_INPUT(GPIOC_ZIO_D16) |        \
                                       PIN_MODE_INPUT(GPIOC_ZIO_D21) |        \
                                       PIN_MODE_INPUT(GPIOC_ZIO_D43) |        \
                                       PIN_MODE_INPUT(GPIOC_ZIO_D44) |        \
                                       PIN_MODE_INPUT(GPIOC_ZIO_D45) |        \
                                       PIN_MODE_INPUT(GPIOC_ZIO_D46) |        \
                                       PIN_MODE_INPUT(GPIOC_ZIO_D47) |        \
                                       PIN_MODE_INPUT(GPIOC_BUTTON) |         \
                                       PIN_MODE_INPUT(GPIOC_OSC32_IN) |       \
                                       PIN_MODE_INPUT(GPIOC_OSC32_OUT))

closed 2025-04-02 by rusefillc