rusefi/rusefi #7669: cranking IAC position should be a curve by engine temerature

rusefillc, 2025-03-31:
what is now a scalar should become a curve of 6 values by engine temperature

[image]

[image]

FDSoftware, 2025-04-01:
We already have a similar functionality currently, with a CLT multiplier at:
`firmware/controllers/actuators/idle_thread.cpp`
```c++
float IdleController::getCrankingOpenLoop(float clt) const {
	float mult =
		interpolate2d(clt, config->cltCrankingCorrBins, config->cltCrankingCorr);

	return engineConfiguration->crankingIACposition * mult;
}
```
TS:
[image]

I think it would be more convenient to recycle this existing thing (ie, removing crankingIACposition and update the curve for use multiplier) is it correct?

rusefillc, 2025-04-01:
@FDSoftware yes!

rusefillc, 2025-04-02:
https://github.com/rusefi/rusefi/pull/7672

closed 2025-04-02 by rusefillc