rusefi/rusefi #7677: electronic wastegate actuator PID anti-windup

adamg-86, 2025-04-04:
Problem: electronic wastegate PID. The I term accumulate a lot of error and is never set back to 0 when a quick change is requested. It makes the control way less effective. 

For example I request 100% actual position of the sensor = 99.5. The I accumulate to +160.
I ask to move to 0 but the I term fight against the P term. 

[image]

[image]

A setting to reset the I term when the P term is big (+ or -) would mostly solve the problem.
And set a max accumulation of the I term would be nice too. 

Thank you good sir.

rusefillc, 2025-04-04:
I assume that https://github.com/rusefi/rusefi/issues/7150 would have to happen first

unfortunately not much priority at the moment

adamg-86, 2025-05-06:
To get back to this. 
The error is mostly cause by the accumulation at the limits. 
If a simple menu to set the effective range of the I term would fix it. 
outside this range the I term is deactivated and set to 0. 

The problem arise mostly to keep pressure on the seat of the wastegate. You have to adjust it at about 5% of the travel is close. That way the close loop is applying force to keep it close to have a shorter spool time. You don't want the I term to accumulate in this position. This cause really bad overboost from the I term error acumulation. Thx god for boost cut 

[image]

[image]