rusefi/rusefi #7681: Downshift ETB Blip

ElDominio, 2025-04-04:
**Is your feature request related to a problem? Please describe.**
 I'm always frustrated when I downshift my ETB car and the RPMs do not rise while downshifting, which would help me shift to a lower gear since synchros are happier the closer to mesh speed they are.

**Would we need new configuration fields?**
There would have to be a new menu under "Advanced" called "Downshift ETB Blip ", containing at least the following settings:
Downshift blip activation = Activated/Deactivated
Downshift Blip Active Input = Always active/Pin activation with pin choice/Lua (allow to be shared with torque reduction trigger pin/clutch input)
Downshift Blip RPM Minimum = RPM minimum to enable blip functionality.
Downshift Blip Time = the time ETB is held at the target ETB Blip table (up to 5000ms)
Downshift Blip Arm APP = we only want to blip if the user has been using APP past a certain threshold, not when cruising about.
Downshift Blip Trigger APP = we want to blip ONLY if the user lets go of the pedal, we don't want to blip is the user stayed with the pedal pressed, this value creates that ceiling
Downshift Blip Timeout Functionality = Enables or overrides Downshift blip Timeout. If Timeout Functionality is disabled, we will only blip ETB once for the amount of time entered in Downshift Blip Time. If the functionality is enabled, we blip for the Downshift Blip Time, wait the Downshift Blip Timeout time, then blip again in an infinite loop until conditions are untrue.
Downshift Blip Timeout = time to wait between possible activation of blip (up to 5000ms)
Downshift Blip ETB Target Table = 6x6 table with RPM vs TPS. This value represents what ETB value the ETB will be held at once blip conditions are met

**Describe the solution you'd like**
once the user crosses over Arm APP AND under Trigger APP AND RPM minimum is passed, the conditions are met, and actual ETB target becomes the value grabbed from **Downshift Blip ETB Target Table** for  **Downshift Blip Time** milliseconds.

**Describe alternatives you've considered**
Should be implemented in Lua and tested before full implementation

mmatousek89, 2025-04-04:
Downshift Blip Arm APP = we only want to blip if the user has been using APP past a certain threshold, not when cruising about.

This make no sense-- unless it's somehow delayed--

Because in real time blip allways happen when your right foot is on brake and APP= 0%

ElDominio, 2025-04-04:
> Downshift Blip Arm APP = we only want to blip if the user has been using APP past a certain threshold, not when cruising about.
> 
> This make no sense-- unless it's somehow delayed--
> 
> Because in real time blip allways happen when your right foot is on brake and APP= 0%

I am pretty sure that in this situation, APP reacts much faster than physical TPS, allowing us to determine if we need to change TPS position.
Having the delay would definitely help, and I totally missed the requirement of either brake input or some calculation of rpm/s to determine we are in a braking action (in motorcycles blip happens even no brake is applied, since engine braking is much more pronounced)