rusefi/rusefi #7686: LUA: !!! ERROR: canTx function is nil !!!

Abricosvw, 2025-04-05:
**Title:** Issue with `canTx` Function Call in Lua Script for Sending CAN Bus Messages

**Description:**
I encountered an issue when trying to send a message via CAN bus using the `canTx` function in a Lua script. The `canTx` function is found to be `nil`, which causes an error when executing the script. Here are the steps to reproduce the issue and the corresponding script:

1. Create a Lua script to send a CAN bus message with identifier `0x0004` and data `00 00 30 75`.
2. Execute the script in the rusEFI environment.

**Expected Behavior:**
The message should be successfully sent via CAN bus.

**Actual Behavior:**
The `canTx` function is found to be `nil`, which causes an error when executing the script.

**Logs:**
```
2025-04-05_07_52_11_939: EngineState: confirmation_luareset:8
2025-04-05_07_52_11_939: EngineState: LUA: Tearing down instance...
2025-04-05_07_52_11_940: EngineState: LUA loading script length: 882...
2025-04-05_07_52_11_940: EngineState: LUA: --- SIMPLE CAN TX SCRIPT ---
2025-04-05_07_52_11_940: EngineState: LUA: !!! ERROR: canTx function is nil !!!
2025-04-05_07_52_11_940: EngineState: LUA script loaded successfully!
2025-04-05_07_52_11_940: EngineState: Lua memory heap usage: 13160 / 60000 bytes = 21.9%
```

**Script:**

```lua
-- Simple script for sending a CAN bus message

print("--- SIMPLE CAN TX SCRIPT ---")

-- --- Configuration ---
local can_bus = 0 -- Using Bus 0
local can_id = 0x0004 -- Message identifier
local data_length = 4 -- Data length
local data = {0x00, 0x00, 0x30, 0x75} -- Message data

-- Check for required functions
if canTx == nil then
    print("!!! ERROR: canTx function is nil !!!")
    return
else
    print("canTx function is available.")
end

if os == nil then
    print("!!! ERROR: os library is nil !!!")
    return
elseif os.delay == nil then
    print("!!! ERROR: os.delay function is nil !!!")
    return
else
    print("os and os.delay functions are available.")
end

print("Starting CAN send...")

-- Send CAN message
local success, err = pcall(function()
    canTx(can_bus, can_id, true, data) -- isExt = true for extended ID
end)

if not success then
    print("!!! ERROR: Failed to send CAN message: " .. tostring(err))
else
    print("CAN message sent successfully!")
end

print("Script finished.")
```

Please help to resolve this issue and provide recommendations for fixing it.

---

rusefillc, 2025-04-05:
@ElDominio check this shit out, our first AI defect report, probably hallucination

Abricosvw, 2025-04-06:
The unusual way the code is written can be misleading. 

txCan(bus, ID, isExt, payload)
canRxAdd(id)
canRxAdd(bus, id)
canRxAdd(id, callback)
canRxAdd(bus, id, callback)
canRxAddMask(id, mask)
canRxAddMask(bus, id, mask)

not  ---      canTx   
correct --   txCan

closed 2025-04-06 by Abricosvw