rusefi/rusefi #7700: Dual bank F7 interrupts execution while erasing/writing flash following "burn" command

ggurov, 2025-04-09:
**Describe the bug**
Dual bank F7, in dual bank mode, interrupts execution while erasing/writing flash after clicking "burn"

**Hardware**
 - Board: F767ZIT6 Nucleo, F767BIT6   - dual bank 2mb MCU
 - Engine info N/A
 - 
**To Reproduce**
Steps to reproduce the behavior:
1. Flash latest appropriate F7 code (f7 proteus,  f7 nucleo)
2. Configure any channel for ignition output
3. Connect Logic analyzer to that ignition output
4. Go to ECU Stimulator
5. Start stimulator
6. Start live capture on logic analyzer
7. Change a value, click burn, causing flash to be written
8. Notice a break in ignition signals, and pause in the serial data coming in.

This will likely stall the engine, as ignition and injection go away for about 2 seconds, leaving channels in whatever state they were:

**Screenshots**

[image]

**Additional context**

This is pretty dangerous behavior as there could be  1-2 seconds of fuel sprayed, and ignition coils may be left charging for 1-2 seconds.

rusefillc, 2025-05-18:
@dron0gus do we have/can we have any official F7 burning in background? if not we need code comments why

rusefillc, 2025-05-18:
@dron0gus see https://github.com/rusefi/rusefi/commit/cccc4cec52ddfcdda920d4d6732a742acc926673

dron0gus, 2025-05-19:
@ggurov I assume you are testing debug build?
`2mb_flash.mk` is enabled for nuclue_f767. This is dangerous because allows up to 1.5Mb for code. 0.5 of this is located in second bank, that is used for settings store. So when settings are writen to second bank any code execution from this high 0.5 Mb will freeze CPU until write/erase is finished.
Changing "-O0" optimization level to "-Og" (still good for debug) significantly reduces code size and it fits in 1Mb.
With this modification we can remove dangerous `2mb_flash.mk` hack.

Also see https://github.com/rusefi/rusefi/issues/6242 https://github.com/rusefi/rusefi/commit/45ac95e2493b4fffdbdd12b38b74024a354c59b8

ggurov, 2025-05-19:
i tried a bunch of different combinations, including the proteus build on build server, on a dual bank f7, that also freezes while writing.

so yeah, proteus f7 code freezes, i tried building my own to add debugging printf around paths, and freeze is around erase part, as expected.

ggurov, 2025-06-17:
just tested dual bank nucleo, it's "better", but there's still definitely a CHUG.  much shorter chug, but it's present.

ggurov, 2025-06-17:
it's now 360 ms, twice, before it was 820ms, twice. engine would still stall.

logic analyzer hooked up to PD13, which is coil1 output:

[image]

[image]

[image]

this is blank dual bank nucleo with :

rusefi_bundle_stm32f767_nucleo.zip  2025-06-16 20:45   61M

engine stim, 1200 rpm, single ignition output

open VE table 
change one value
click burn

AND THIS

[image]

is changing a few VE table values,  NOT clicking burn, but JUST clicking CLOSE.

[image]

[image]

[image]

same setup, same session, same engine stim from before still going.

"settings pending" indicator briefly flashes as burn is requested, and turns off, after all the signal drama is over.