rusefi/rusefi #7717: pt2001 timeout to trigger warning for pump

rusefillc, 2025-04-14:
we have ``valve maximum duration`` do we check against it?

[image]

FDSoftware, 2025-04-15:
After checking the repo,

* mc33_hpfp_max_hold is being sent to the PT2001 in the setup code
* In the McFault enum, we don't have a flag to know if an error of this type occurred
* readDriverStatus seems to only run when *configuring* the PT2001 or when a reboot occurs. Is it very expensive to call this function within the limp manager to see if a fault occurred while the engine is running? (I see that it has to communicate via SPI and it takes some time) [I also assume that some of the PT2001's microcode would have to be modified to add this new fault]
* I don't see that we have LiveData for this?

Since points 2-3 are a bit beyond what I can do, can I add a var inside `high_pressure_fuel_pump_s` and then use it within the limp manager? (it won't work because point 2-3 would be missing, but to be able to move things forward)

rusefillc, 2025-04-15:
@FDSoftware i was focused on the high_pressure_fuel_pump side of things where we command how long we pump to be open

same as with GDI - we did not look into GDI side, we looked into scheduling part

FDSoftware, 2025-04-15:
Ok cool, so by checking the deadtime + hpfpActivationAngle I can now do the check

closed 2025-04-15 by rusefillc