rusefi/rusefi #7725: Hemi 392 Cam VVT Decoder

ElDominio, 2025-04-15:
**Is your feature request related to a problem? Please describe.**
I would like to have full sequential fuel on my Durango 392, running on PNP converted Platinum

**Describe the solution you'd like**
Add support for this trigger

**Additional context**

[392log.zip](https://github.com/user-attachments/files/19766716/392log.zip)
Logic log, valid data is from 4 to 5 seconds

rusefillc, 2025-04-15:
@ElDominio any photos from the internet showing genetal shape and tooth count?

ElDominio, 2025-04-15:
[image]

This is the most info i can get from Haltech

rusefillc, 2025-04-15:
let me request $100 for https://www.ebay.com/itm/286122743488

rusefillc, 2025-04-15:
[image]

[image]

ElDominio, 2025-04-16:
[image]

This is crank BTW; I have no idea why Logic shows odd things with cam teeth... I think I did not ground the logic analyzer which probably introduced noise?

ElDominio, 2025-04-16:
I did not ground the logic analyzer when i recorded the logicdata log, my bad

closed 2025-04-16 by rusefillc

ElDominio, 2025-04-16:
I will do it correctly tomorrow