rusefi/rusefi #7745: Feature Request: Toyota 1UZ,2UZ,3UZ VVT Decoder

AKCore, 2025-04-19:
Toyota 3 Tooth does not work for 1UZ VVTi, and newer UZ platform VVT configurations.
The 3 tooth trigger is an offset trigger unlike the 2jz, 

Attached is both a trigger log directly from the VR pickups and also digital filtered through a max9926

Raw VR scoped. 
Ch1 Crank, Ch3 Bank 1 VVT, Ch2, Bank 2 VVT
[image]

[image]

[image]

This is digital filtered max, Cam Bank 1 is CH2, Cam Bank 2 is CH3 and Crank is CH1
[image]

rusefillc, 2025-04-19:
@AKCore scope photos provide very little useful info

Please provide a photo of actual physical cam wheel

AKCore, 2025-04-19:
I don't really have one, I have this one out of a youtube video 
From what i understand its, 135* gap, 90* gap, then 135* gap.
Without removing the cams, i don't think i can get you a good picture of the camshaft.  

[image]

AKCore, 2025-04-23:
is there anything else you need from me for this enhancement? Can i give you some dollars?

rusefillc, 2025-04-24:
> is there anything else you need from me for this enhancement? Can i give you some dollars?

I wish it was that easy unless we are talking a five digit number

commit 533f2b1a (2025-04-24): Feature Request: Toyota 1UZ,2UZ,3UZ VVT Decoder hopefully fix #7745