rusefi/rusefi #7747: unit test make is not incremental

rusefillc, 2025-04-20:
Have we broken incremental compilation again?

cd unit_tests
make -j12
make -j12

expected: second invocation is incremental/not doing much
actual: full re-compilation

rusefillc, 2025-04-20:
see also https://github.com/rusefi/rusefi/issues/7748

FDSoftware, 2025-04-20:
last good commit: f7cee15f47de493aeb5479f9e17f8442047366f4
first bad commit: 1790ff263eb239c1fce6f55b505cf6a445c29238
related? issue: https://github.com/rusefi/rusefi/issues/6711

rusefillc, 2025-04-20:
https://github.com/rusefi/rusefi/issues/6082

rusefillc, 2025-04-20:
```
user@dell1:~/rusefi/unit_tests$ make -d > a
 grep -v Reading a | grep -v Trying | grep -v Looking | grep -v Rejecting | grep -v Avoiding | grep -v 'No need' | grep -v 'No im' | grep -v Considering | grep -i newer | less
```

rusefillc, 2025-04-20:
```
         Prerequisite '../firmware/console/binary/generated/live_data_fragments.ini' is newer than target '.config-sentinel'.
         Prerequisite '../firmware/console/binary/generated/data_logs.ini' is newer than target '.config-sentinel'.
       Prerequisite '.config-sentinel' is newer than target '../firmware/tunerstudio/generated/rusefi_f407-discovery.ini'.
       Prerequisite '.docsenums-sentinel' is newer than target '../firmware/tunerstudio/generated/rusefi_f407-discovery.ini'.
       Prerequisite '.config-sentinel' is newer than target '../firmware/controllers/generated/rusefi_generated_f407-discovery.h'.
       Prerequisite '.docsenums-sentinel' is newer than target '../firmware/controllers/generated/rusefi_generated_f407-discovery.h'.
       Prerequisite '.config-sentinel' is newer than target '../firmware/controllers/generated/signature_f407-discovery.h'.
       Prerequisite '.docsenums-sentinel' is newer than target '../firmware/controllers/generated/signature_f407-discovery.h'.
       Prerequisite '.config-sentinel' is newer than target '../firmware/controllers/generated/engine_configuration_generated_structures_f407-discovery.h'.
       Prerequisite '.docsenums-sentinel' is newer than target '../firmware/controllers/generated/engine_configuration_generated_structures_f407-discovery.h'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/test_buffered_writer.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/test_error_accumulator.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/test_exp_average.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/test_honda_crc.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/test_closed_loop_controller.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/test_scaled_channel.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/test_timer.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/test_averaging.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/test_lua_biquad.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/test_hash.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/lua.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/lua_hooks.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/can_filter.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/lua_hooks_util.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/script_impl.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/lua_can_rx.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/output_lookup_generated.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/value_lookup_generated.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/state_sequence.o'.
   Prerequisite '../firmware/pch/pch.h.gch/unit_tests' is newer than target 'build/obj/big_buffer.o'.
```

commit 47d8c677 (2025-04-20): unit test make is not incremental fix #7747
Fields.java is a mess #6711
closed 2025-04-20 by rusefillc