rusefi/rusefi #7748: CI coverage to assert that unit test build is incremental

rusefillc, 2025-04-20:
we need to protect incremental compilation since we broke it more than once

rusefillc, 2025-04-20:
i wonder how can we CI this?

time second invocation of make maybe?

```
user@dell1:~/rusefi/unit_tests$ make -j12 -d | grep 'is newer'
       Prerequisite '.config-sentinel' is newer than target '../firmware/tunerstudio/generated/rusefi_f407-discovery.ini'.
       Prerequisite '.docsenums-sentinel' is newer than target '../firmware/tunerstudio/generated/rusefi_f407-discovery.ini'.
       Prerequisite '.config-sentinel' is newer than target '../firmware/controllers/generated/rusefi_generated_f407-discovery.h'.
       Prerequisite '.docsenums-sentinel' is newer than target '../firmware/controllers/generated/rusefi_generated_f407-discovery.h'.
       Prerequisite '.config-sentinel' is newer than target '../firmware/controllers/generated/signature_f407-discovery.h'.
       Prerequisite '.docsenums-sentinel' is newer than target '../firmware/controllers/generated/signature_f407-discovery.h'.
       Prerequisite '.config-sentinel' is newer than target '../firmware/controllers/generated/engine_configuration_generated_structures_f407-discovery.h'.
       Prerequisite '.docsenums-sentinel' is newer than target '../firmware/controllers/generated/engine_configuration_generated_structures_f407-discovery.h'.
 Prerequisite 'build/obj/lapi.o' is newer than target 'all'.
 Prerequisite 'build/obj/lcode.o' is newer than target 'all'.
 Prerequisite 'build/obj/lctype.o' is newer than target 'all'.
 Prerequisite 'build/obj/ldebug.o' is newer than target 'all'.
 Prerequisite 'build/obj/ldo.o' is newer than target 'all'.
 Prerequisite 'build/obj/ldump.o' is newer than target 'all'.
 Prerequisite 'build/obj/lfunc.o' is newer than target 'all'.
 Prerequisite 'build/obj/lgc.o' is newer than target 'all'.
 Prerequisite 'build/obj/llex.o' is newer than target 'all'.
```

FDSoftware, 2025-04-20:
> [@FDSoftware](https://github.com/FDSoftware) [359ed83](https://github.com/rusefi/rusefi/commit/359ed83a3194e87b4fe7502578a3c5c28e447a84) we can declare incremental if less than 30 seconds?

I think so, locally it takes 2-3s to rebuild, full build 3m

FDSoftware, 2025-04-20:
also some tests from @esaulenka about build time here: https://github.com/rusefi/rusefi/issues/7507

closed 2025-04-20 by rusefillc

FDSoftware, 2025-04-24:
@rusefillc one real world example of this check working!:
https://github.com/FDSoftware/rusefi/actions/runs/14653201987/job/41123438154

```
Compiling f136.cpp
Compiling mazda_miata.cpp
Compiling mazda_miata_base_maps.cpp
Compiling mazda_miata_1_6.cpp
Compiling mazda_miata_na8.cpp
Compiling mazda_miata_vvt.cpp
make: *** [../firmware/../unit_tests/rules.mk:113: build/obj/mazda_miata_base_maps.o] Terminated
Error: Process completed with exit code 124.
```