rusefi/rusefi #7755: can assign o2 input to a duplicate channel

rusefillc, 2025-04-20:
configure AFR and TPS for same input

expected: error
actual: no error even after power cycle

[image]

rusefillc, 2025-04-20:
@ggurov says broken somewhere around https://github.com/rusefi/rusefi/commit/86c0fcdd03d46c6df8b64a31d4e3923ece9946e6 maybe #7718

ggurov, 2025-04-20:
no error here already:

[Lambda1/Lambda2 exp average smoothing [preparation] see #7717](https://github.com/rusefi/rusefi/actions/runs/14475096621)

i

ggurov, 2025-04-20:
error comes back here TS UI: ETB/EWG bias curve position axis range 0..100 #30130

therefore [Lambda1/Lambda2 exp average smoothing [preparation] see #7717](https://github.com/rusefi/rusefi/actions/runs/14475096621) must be the b0rk