rusefi/rusefi #7761: migrate FAST_CALLBACK_PERIOD_MS SLOW_CALLBACK_PERIOD_MS to main_loop.h

rusefillc, 2025-04-21:
See https://github.com/rusefi/rusefi/commit/e7b1bddc1ec2feb2d06742c13074962260107082 would that also change from 200Hz to 250hz?

FDSoftware, 2025-04-21:
> See [e7b1bdd](https://github.com/rusefi/rusefi/commit/e7b1bddc1ec2feb2d06742c13074962260107082) would that also change from 200Hz to 250hz?

I don't think the HPFP.PI test likes the change from 200 to 250, it uses the period to obtain the expected result

FDSoftware, 2025-04-21:
draft here adding 200hz as option & using it on the fast callback
https://github.com/rusefi/rusefi/pull/7763

rusefillc, 2025-04-21:
@FDSoftware please https://github.com/rusefi/rusefi/issues/7764

FDSoftware, 2025-04-22:
@rusefillc pls close this & next small step should be migrate SlowAdcController to new main loop?

rusefillc, 2025-04-22:
Let's have @dron0gus drive anything ADC

Can we mive with not ADC or are we depending?

FDSoftware, 2025-04-22:
> Let's have [@dron0gus](https://github.com/dron0gus) drive anything ADC
> 
> Can we mive with not ADC or are we depending?

etb & slow callback & fastcallback, but they are more critical, can I add the empty function and a placeholder for the calls? It wouldn't impact anything.

FDSoftware, 2025-04-23:
> > Let's have [@dron0gus](https://github.com/dron0gus) drive anything ADC
> > Can we mive with not ADC or are we depending?
> 
> etb & slow callback & fastcallback, but they are more critical, can I add the empty function and a placeholder for the calls? It wouldn't impact anything.

done at: https://github.com/rusefi/rusefi/pull/7771

[this is only for easy tracking if someone visits this issue in the future]