rusefi/rusefi #7776: maybe a defect around pt2001 timeout to trigger warning for pump

rusefillc, 2025-04-24:
#7717 

```
2025-04-24_11_40_54_873: EngineState: Time is changed to
2025-04-24_11_40_54_875: EngineState: Current RTC time: 24 Apr 2025 11:40:54
2025-04-24_11_40_54_876: EngineState: confirmation_set date 2025-04-24T11:40:54.3849644:36
2025-04-24_11_41_02_495: EngineState: WARNING: C9013: Injection HPFP pump time excess PT2001 limits time: 31226.1699
2025-04-24_11_41_04_582: EngineState: WARNING: C9002: primary trigger error: too many teeth between sync points: expected 58/0 got 58/0
2025-04-24_11_41_05_247: EngineState: WARNING: C9003: primary trigger error: not enough teeth between sync points: expected 58/0 got 12/0
2025-04-24_11_41_05_654: EngineState: WARNING: C9004: cam VVT B1 Int trigger error: too many teeth between sync points: 8 > 8
2025-04-24_11_41_05_754: EngineState: [err] VVT1 rpm=211 time=274 eventIndex=9 gapIndex=0: n gap=0.436 expected from 0.483 to 0.806 error=Yes
2025-04-24_11_41_05_755: EngineState: [err] VVT1 rpm=211 time=274 eventIndex=9 gapIndex=1: n gap=0.459 expected from 1.167 to 1.944 error=Yes
2025-04-24_11_41_05_866: EngineState: [err] VVT1 rpm=211 time=274 eventIndex=11 gapIndex=0: n gap=0.280 expected from 0.483 to 0.806 error=Yes
```

rusefillc, 2025-04-24:
@FDSoftware definitely a defect

```engine->module<HpfpController>().unmock().m_deadangle + static_cast<angle_t>(engineConfiguration->hpfpActivationAngle);```

m_deadangle is dead time - that's part of duration
hpfpActivationAngle configuration is POSITION, not duration? i.e. has nothing to do with duration check?

FDSoftware, 2025-04-24:
> m_deadangle is dead time - that's part of duration

comment on `m_deadangle` says `///< Computed solenoid deadtime in degrees`
for dead time is `deadtime_ms` I assume?
dead angle math is:

`m_deadangle = deadtime_ms * rpm * (360.f / 60.f / 1000.f);`

> hpfpActivationAngle configuration is POSITION, not duration? i.e. has nothing to do with duration check?

rusefi.txt says:

`uint8_t hpfpActivationAngle;How long to keep the valve activated (in order to allow the pump to build pressure and keep the valve open on its own)`
maybe we need to change this text?

also on live data we have more real:
```c++
angle_t m_requested_pump;GDI: HPFP duration\nComputed requested pump duration in degrees (not including deadtime)
```

so correct check should be:

```c++
angle_t finalHpfpPumpAngle = engine->module<HpfpController>().unmock().m_deadangle + engine->module<HpfpController>().unmock().m_requested_pump
```

is this correct in this way?

rusefillc, 2025-04-24:
@FDSoftware I am just a carousel horse but here is more confusion

```
void HpfpController::pinTurnOn(HpfpController *self) {
	enginePins.hpfpValve.setHigh(HPFP_CONTROLLER);
	self->HpfpValveState = true;
	self->HpfdActivationPhase = getTriggerCentral()->getCurrentEnginePhase(getTimeNowNt()).value_or(-1);

	// By scheduling the close after we already open, we don't have to worry if the engine
	// stops, the valve will be turned off in a certain amount of time regardless.
	scheduleByAngle(&self->m_event.eventScheduling,
			self->m_event.eventScheduling.getMomentNt(),
			self->m_deadangle + engineConfiguration->hpfpActivationAngle,
			{ pinTurnOff, self });
}
```

rusefillc, 2025-04-24:
^^^ ohh that code is legit - that's emergency stop

rusefillc, 2025-05-18:
this was a bad idea, pump control is different from say injector control - we only care to open value with remaining magic being mechanical not electrical

closed 2025-05-18 by rusefillc