rusefi/rusefi #7778: what is hpfpActivationAngle

rusefillc, 2025-04-24:
So what is hpfpActivationAngle?

on the one hand we have ``How long to keep the valve activated (in order to allow the pump to build pressure and keep the valve open on its own)`` tooltip

on the other hand the only .cpp usage is 

```
	// By scheduling the close after we already open, we don't have to worry if the engine
	// stops, the valve will be turned off in a certain amount of time regardless.
	scheduleByAngle(&self->m_event.eventScheduling,
			self->m_event.eventScheduling.getMomentNt(),
			self->m_deadangle + engineConfiguration->hpfpActivationAngle,
			{ pinTurnOff, self });
```

rusefillc, 2025-04-24:
@FDSoftware does it look like hpfpActivationAngle is only "activation limit for engine stop"? if that's the case we might not need it configurable at all?

FDSoftware, 2025-04-24:
> [@FDSoftware](https://github.com/FDSoftware) does it look like hpfpActivationAngle is only "activation limit for engine stop"? if that's the case we might not need it configurable at all?

It seems that "m_requested_pump" was missing there on `pinTurnOn`, also the comment doesn't seem to be entirely accurate, since after `pinTurnOff`, `scheduleNextCycle` is called again, so the pin can be activated again if the calculated angle is correct, even if the motor is stopped!

```c++
void HpfpController::pinTurnOff(HpfpController *self) {
	enginePins.hpfpValve.setLow(HPFP_CONTROLLER);
	self->HpfpValveState = false;
	self->HpfdDeactivationPhase = getTriggerCentral()->getCurrentEnginePhase(getTimeNowNt()).value_or(-1);

	self->scheduleNextCycle();
}
`

rusefillc, 2025-04-24:
@FDSoftware I have no idea what are you saying. Are you responding to original question or raising g one or two additional issues?

FDSoftware, 2025-04-24:
> or raising g one or two additional issues?

No, I was just confused about how `HpfpController::scheduleNextCycle()` works, maybe a comment like this would help the next person who reads the code:

```c++
	// m_requested_pump will be 0 in case of !isHpfpActive, read "HpfpController::onFastCallback()" for the logic
	angleAboveMin = angle_requested > engineConfiguration->hpfpMinAngle;
	if (angleAboveMin) {
		// TODO: some manuals suggest also substracting peak time (converted to angle)
```

now, about: 
> So what is hpfpActivationAngle?

"hpfpActivationAngle" is the hardcoded duration at which the valve is active, now, at the same time we have:

`angle_t m_requested_pump;GDI: HPFP duration\nComputed requested pump duration in degrees (not including deadtime)`

calculated by:

```c++
angle_t HpfpQuantity::pumpAngleFuel(float rpm, HpfpController *model) {
	// Math based on fuel requested
	model->fuel_requested_percent = calcFuelPercent(rpm);

	model->fuel_requested_percent_pi = calcPI(rpm, model->fuel_requested_percent, model);
	// Apply PI control
	float fuel_requested_percentTotal = model->fuel_requested_percent + model->fuel_requested_percent_pi;

	// Convert to degrees
	return interpolate2d(fuel_requested_percentTotal,
			config->hpfpLobeProfileQuantityBins,
			config->hpfpLobeProfileAngle);
}
```
For now, it is only used by scheduleNextCycle to know whether it should be activated or not.

which leads me to think that `hpfpActivationAngle` should be replaced by `m_requested_pump`

The only detail is that the comment on pumpAngleFuel adds a bit of general confusion saying that it is the angle where it is activated and not the duration in angles.

```c++
class HpfpQuantity {
public:
	/**
	 * Calculate where the pump should become active, in degrees before pump lobe TDC
	 */
	angle_t pumpAngleFuel(float rpm, HpfpController *model);

	/**
	 * Calculate the percent of the pump stroke needed to replace the fuel injected.  Also
	 * includes pump compensation calculations.
	 *
	 * This is used by internal tests and shouldn't be called directly.  Instead use
	 * pumpAngleFuel.
	 *
	 * Return value is nominally 0-100, but may be outside that range (including negative) if
	 * model parameters are not accurate.
	 */
	float calcFuelPercent(float rpm);
```

rusefillc, 2025-04-24:
> "hpfpActivationAngle" is the hardcoded duration at which the valve is active, now, at the same time we have:

No?

Please note that we have two references to pinTurnOff

So, the flow is

* schedule pinTurnOn with emergency hpfpActivationAngle shutoff (first usage of pinTurnOff)(
* also schedule proper pinTurnOff based on lobeAngle

FDSoftware, 2025-04-24:
> * also schedule proper pinTurnOff based on lobeAngle

second reference to pinTurnOff says is for re-schedule in case of non-valid pump angle :c

```
	// Schedule this, even if we aren't opening the valve this time, since this
		// will schedule the next lobe.
		// todo: would it have been cleaner to schedule 'scheduleNextCycle' directly?
		engine->module<TriggerScheduler>()->schedule(
			HPFP_CONTROLLER,
			&m_event, lobeAngle,
			{ pinTurnOff, this });
```

rusefillc, 2025-04-24:
TL,DR: we are somewhat confused, I've added a code comment to document that we are confused :(

@racer-coder FYI

closed 2025-04-24 by rusefillc

racer-coder, 2025-04-24:
You committed code that removed the description of what hpfpActivationAngle does in favor of saying "I dunno, we're confused."
I thought the code was pretty self explanatory - it turns the pin off deadtime_angle + hpfpActivateAngle degrees later after turning the pin on, right?
Yes if we don't activate the pump on this lobe (not cycle! there are multiple lobes per cycle) then I just call pinoff to wake up the code a bit later to see if it's relevant for the next lobe.

Have you read https://rusefi.com/forum/viewtopic.php?f=5&t=2192
These comments might help:
For the solenoid type pump, the pump has a certain volume. You can activate the solenoid to request that the pump start pressurizing. Once it reaches a certain pressure, it is effectively self running and won't unlatch until the pump reaches the top. 
Since the solenoid latches itself, you don't have to keep it activated for the whole lobe. You just need to activate it until it latches and then let it do the rest of the work.

rusefillc, 2025-04-24:
@racer-coder thank you for the explanation! No, I was not getting this :( See https://github.com/rusefi/rusefi/commit/574a8ea54156f3a2d100f7d637452b80256a6b13

Wider context: We've attempted to add (naive?) validation of duration against ASIC microcode timeout https://github.com/rusefi/rusefi/issues/7717 and that did not work great https://github.com/rusefi/rusefi/issues/7776

rusefillc, 2025-04-24:
@racer-coder you read it correct - I have in fact killed the tooltip earlier today while misunderstanding it. I've revered with hopefully improvements based on your clarification.