rusefi/rusefi #7785: Sync:Known engine phase is not working

ElDominio, 2025-04-25:
**Describe the bug**
No angle is reported on Sync: Known Engine Phase

**Hardware**
 - Board: AlphaX Gold revA
 - Engine info 2GRFE

**To Reproduce**
Steps to reproduce the behavior:
do a log
check known enigne phase
no engine phase

**Expected behavior**
I have an angle for engine phase

**Additional context**

[2025-04-24_20.05.02.mlg.zip](https://github.com/user-attachments/files/19900666/2025-04-24_20.05.02.mlg.zip)

rusefillc, 2025-04-25:
@ElDominio please clarify 

[image]

rusefillc, 2025-04-25:
@ElDominio better snippet

[image]

ElDominio, 2025-04-25:
Need to clarify somewhere that _KNOWN_ engine phase != engine phase without knowing

commit bbf0166a (2025-04-25): Sync:Known engine phase is not working fix #7785
closed 2025-04-25 by rusefillc