rusefi/rusefi #7793: jumpy WBO

rusefillc, 2025-04-25:
is rusEFI broken or WBO broken or car sensor broken?

[image]

[wbo-issue.zip](https://github.com/user-attachments/files/19917265/wbo-issue.zip)

dron0gus, 2025-04-28:
@rusefillc 
Sensor falls to
`#define HACK_SILENT_VALUE 1`
state due to unknown reason. AEM/RusEFI WBO timeout is 30 mS, while default AFR messages rate is 100Hz (10 mS).
Do you have high CAN bus load?
While it is in timeout/silent error state ECU still receives updated data:

[image]
This is possible as part of data is transfered with another CANID. And only primary CANID message resets timeout counter.

rusefillc, 2025-04-30:
CANbus trace file

[wbo-issue-trace.zip](https://github.com/user-attachments/files/19969448/wbo-issue-trace.zip)

rusefillc, 2025-05-01:
@dron0gus 

```
2025-05-01_11_33_06_244: EngineState: [verbosecan] enabled
2025-05-01_11_33_06_246: EngineState: confirmation_enable verbosecan:17
2025-05-01_11_33_06_247: EngineState: BENCH_TEST Sending CAN1 message: ID=770000/l=8 23 0 44 0 0 0 0 0
2025-05-01_11_33_06_248: EngineState: BENCH_TEST Sending CAN1 message: ID=770001/l=8 BA 44 31 19 2A 44 52 65
2025-05-01_11_33_06_249: EngineState: BENCH_TEST Sending CAN1 message: ID=770006/l=8 0 0 0 0 0 0 3D 0
2025-05-01_11_33_06_250: EngineState: WBO_SERVICE Sending CAN1 message: ID=EF50000/l=2 74 1 0 0 0 0 0 0
2025-05-01_11_33_06_250: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 F3 36 0B 03 00 00
2025-05-01_11_33_06_251: EngineState: CAN1 RX: ID 191(401) DLC 8: 2C 01 56 01 00 00 AF 00
2025-05-01_11_33_06_252: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 D5 3F 0B 03 00 00
2025-05-01_11_33_06_253: EngineState: CAN1 RX: ID 191(401) DLC 8: 2C 01 9A 02 FF 00 AF 00
2025-05-01_11_33_06_255: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 50 30 0C 03 00 00
2025-05-01_11_33_06_256: EngineState: CAN1 RX: ID 191(401) DLC 8: 2C 01 FA 00 00 00 AF 00
2025-05-01_11_33_06_257: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 86 34 0B 03 00 00
2025-05-01_11_33_06_257: EngineState: CAN1 RX: ID 191(401) DLC 8: 2B 01 4D 02 8D 00 AF 00
2025-05-01_11_33_06_258: EngineState: WBO_SERVICE Sending CAN1 message: ID=EF50000/l=2 74 1 0 0 0 0 0 0
2025-05-01_11_33_06_259: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 7A 33 0C 03 00 00
2025-05-01_11_33_06_260: EngineState: CAN1 RX: ID 191(401) DLC 8: 2B 01 3A 01 00 00 AF 00
2025-05-01_11_33_06_260: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 01 34 3A 0B 03 00 00
2025-05-01_11_33_06_261: EngineState: CAN1 RX: ID 191(401) DLC 8: 2C 01 6E 01 00 00 A4 00
2025-05-01_11_33_06_262: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 1D 3A 0B 03 00 00
2025-05-01_11_33_06_263: EngineState: CAN1 RX: ID 191(401) DLC 8: 2C 01 7D 02 FF 00 A4 00
2025-05-01_11_33_06_263: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 9E 44 0B 03 00 00
2025-05-01_11_33_06_264: EngineState: CAN1 RX: ID 191(401) DLC 8: 2C 01 B0 02 FF 00 A4 00
2025-05-01_11_33_06_265: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 E7 2B 0C 03 00 00
2025-05-01_11_33_06_266: EngineState: CAN1 RX: ID 191(401) DLC 8: 2B 01 F0 00 00 00 A4 00
2025-05-01_11_33_06_336: EngineState: BENCH_TEST Sending CAN1 message: ID=770000/l=8 A3 0 4E 0 0 0 0 0
2025-05-01_11_33_06_338: EngineState: BENCH_TEST Sending CAN1 message: ID=770001/l=8 BA 44 31 19 2A 44 52 65
2025-05-01_11_33_06_339: EngineState: BENCH_TEST Sending CAN1 message: ID=770006/l=8 0 0 0 0 0 0 3D 0
2025-05-01_11_33_06_340: EngineState: BENCH_TEST Sending CAN1 message: ID=770005/l=8 66 0 12 E 1 0 0 0
2025-05-01_11_33_06_341: EngineState: BENCH_TEST Sending CAN1 message: ID=770003/l=8 1 37 0 0 EE 0 41 0
2025-05-01_11_33_06_342: EngineState: BENCH_TEST Sending CAN1 message: ID=770004/l=8 0 0 0 0 0 0 0 0
2025-05-01_11_33_06_344: EngineState: BENCH_TEST Sending CAN1 message: ID=770008/l=8 0 0 0 0 0 0 0 0
2025-05-01_11_33_06_345: EngineState: WBO_SERVICE Sending CAN1 message: ID=EF50000/l=2 74 1 0 0 0 0 0 0
2025-05-01_11_33_06_346: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 AD 2F 0B 03 00 00
2025-05-01_11_33_06_346: EngineState: CAN1 RX: ID 191(401) DLC 8: 2C 01 33 01 00 00 A4 00
2025-05-01_11_33_06_347: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 96 3A 0B 03 00 00
2025-05-01_11_33_06_348: EngineState: CAN1 RX: ID 191(401) DLC 8: 2C 01 88 02 FF 00 A4 00
2025-05-01_11_33_06_349: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 DC 30 0B 03 00 00
2025-05-01_11_33_06_350: EngineState: CAN1 RX: ID 191(401) DLC 8: 2C 01 FD 00 00 00 A4 00
2025-05-01_11_33_06_350: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 A3 38 0B 03 00 00
2025-05-01_11_33_06_351: EngineState: CAN1 RX: ID 191(401) DLC 8: 2C 01 7F 02 FF 00 A4 00
2025-05-01_11_33_06_360: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 32 42 0B 03 00 00
2025-05-01_11_33_06_363: EngineState: CAN1 RX: ID 191(401) DLC 8: 2C 01 A3 02 FF 00 A4 00
2025-05-01_11_33_06_363: EngineState: WBO_SERVICE Sending CAN1 message: ID=EF50000/l=2 73 1 0 0 0 0 0 0
2025-05-01_11_33_06_364: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 62 2E 0B 03 00 00
2025-05-01_11_33_06_369: EngineState: CAN1 RX: ID 191(401) DLC 8: 2C 01 F4 00 00 00 B6 00
2025-05-01_11_33_06_369: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 34 37 0B 03 00 00
2025-05-01_11_33_06_370: EngineState: CAN1 RX: ID 191(401) DLC 8: 2C 01 63 02 DF 00 B6 00
2025-05-01_11_33_06_376: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 75 2F 0C 03 00 00
2025-05-01_11_33_06_376: EngineState: CAN1 RX: ID 191(401) DLC 8: 2B 01 F5 00 00 00 B6 00
2025-05-01_11_33_06_380: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 34 37 0B 03 00 00
2025-05-01_11_33_06_383: EngineState: CAN1 RX: ID 191(401) DLC 8: 2B 01 6B 02 BB 00 B6 00
2025-05-01_11_33_06_386: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 7C 33 0C 03 00 00
2025-05-01_11_33_06_389: EngineState: CAN1 RX: ID 191(401) DLC 8: 2B 01 53 01 00 00 B6 00
2025-05-01_11_33_06_392: EngineState: BENCH_TEST Sending CAN1 message: ID=770000/l=8 27 0 56 0 0 0 0 0
2025-05-01_11_33_06_393: EngineState: BENCH_TEST Sending CAN1 message: ID=770001/l=8 BA 44 31 19 29 43 52 65
2025-05-01_11_33_06_395: EngineState: BENCH_TEST Sending CAN1 message: ID=770006/l=8 0 0 0 0 0 0 3D 0
2025-05-01_11_33_06_397: EngineState: WBO_SERVICE Sending CAN1 message: ID=EF50000/l=2 74 1 0 0 0 0 0 0
[... cut ...]
2025-05-01_11_33_07_664: EngineState: BENCH_TEST Sending CAN1 message: ID=770006/l=8 0 0 0 0 0 0 3E 0
2025-05-01_11_33_07_665: EngineState: WBO_SERVICE Sending CAN1 message: ID=EF50000/l=2 74 1 0 0 0 0 0 0
2025-05-01_11_33_07_665: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 8C 35 0C 03 00 00
2025-05-01_11_33_07_665: EngineState: CAN1 RX: ID 191(401) DLC 8: 2B 01 38 01 00 00 98 00
2025-05-01_11_33_07_666: EngineState: CAN1 RX: ID 190(400) DLC 8: A0 00 3B 35 0C 03 00 00
2025-05-01_11_33_07_694: EngineState: CAN1 RX: ID 191(401) DLC 8: 2B 01 19 01 00 00 9F 00

```

rusefillc, 2025-05-05:
@dron0gus i have a second 4K board which seems to be doing exactly same thing.

same ECU patch cord, same WBO cable, different PCBs, one runs late April, the other one runs today binary

dron0gus, 2025-05-05:
Please run following test:
1. enable force heater
2. start logging
3. enable ignition
4. wait 1 min
5. start engine
6. wait few sec
7. stop logging

rusefillc, 2025-05-05:
[WBO.zip](https://github.com/user-attachments/files/20046040/WBO.zip) @dron0gus

dron0gus, 2025-05-06:
1. Выглядит как плохая/уставшая лямбда - Nernst выходит за допустимые пределы, сенсор сбрасывает бит Valid
2. Бага в прошивке - если приходит пакет со сброшенным valid битом - мы не обнуляем тайм-аут и через 30мС перетираем действительную ошибку ошибкой таймаута. Это сбивает с толку.

rusefillc, 2025-05-06:
@dron0gus давай чинить сначала 2? :)

насколько выходит за пределы? можно мы чуть-чуть их расширим хотя бы в тестовой ветке? знаем ли мы, насколько выходит? я с радостью поменяю сенсор, но я пока к нему подступиться не могу :(

dron0gus, 2025-05-06:
@rusefillc 
Да, как раз смотрю на №2.
таргет - 0.45 В +/- 0.1В https://github.com/mck1117/wideband/blob/5c20f128e86bdc9c5d7b5a182f3ab10287dee4f9/firmware/can.cpp#L167
А прыгает от 0.23 до 0.61 на заглушенной и 0.24 .. 074 на заведенной.

Расширить можно. Можно даже перенести логику определния валидности показаний на сторону RE и вынести эти параметры в настройки. Но текущая прошивка WBO зануляет поле лямбды, если внутри себя считает ее невалидной. Логично, но не удобно.

dron0gus, 2025-05-06:
Ксати, возможно у нас не очень оптимальные настройки PID регулятора Ipump. И на этой (уставшей, неоригинальной, новой ревизии?) лямбде у нас осциляции.

FDSoftware, 2025-05-06:
> Ксати, возможно у нас не очень оптимальные настройки PID регулятора Ipump. И на этой (уставшей, неоригинальной, новой ревизии?) лямбде у нас осциляции.

In addition to the PID thing, according to this expired patent and the Waltech blog, the heater's PWM can affect Ipump with noise.
Could we edit the FW to turn off the PWM while the measurement is being performed? I think it could help more than modifying the limits of what Nernst values ​​are valid.

https://patents.google.com/patent/US6712054B2/en

https://waltech.com/cszcms/wide-band-oxygen-sensor-project-report

dron0gus, 2025-05-06:
@FDSoftware WBO is continuously doing Nernst measurement. Then DC and AC part are extracted. DC parts is also filtered/averaged. So no any PWM frequency noise should pass this filter.
Switching off and on pwm, syncing ADC to this and so on will need a lot of rework.
I'll check Nernst signal with oscilloscope to see if it has any noise aligned with heater PWM signal.
Problem is that I'm not able to reproduce this issue on my side.

rusefillc, 2025-05-07:
@mck1117 have you seen this behavior?

rusefillc, 2025-05-07:
customer states

```
So I got the afr reading to stop flickering. I got a 3rd sensor. Doesn’t do it. 
2 sensors flicker on your onboard controller but they work fine with an AEM controller. Very strange but it’s a solution
```

rusefillc, 2025-05-07:
and motec LTC is happy to handle my sensor

FDSoftware, 2025-05-08:
>  I'll check Nernst signal with oscilloscope to see if it has any noise aligned with heater PWM signal

That would be ideal if it isn't too much of a problem for you.

> . Problem is that I'm not able to reproduce this issue on my side.

@rusefillc maybe ask the user for the "defective" sensor or ask for the exact part number? and send to dron for testing?

rusefillc, 2025-05-08:
@FDSoftware yes that's totally the plan! Planning to accumulate some sad/tired sensors