rusefi/rusefi #7830: firmware binaries to have detailed fine names

rusefillc, 2025-05-03:
within both bundle.zip and autoupdate.zip what used to be rusefi.bin will now be called rusefi_development_34234234234.bin or rusefi_lts-25jersey_34234234234_uaefi.bin 

* rusefi_ prefix
* development or lts branch name in case of for instance https://github.com/rusefi/rusefi/tree/lts-25jersey
* git hash
* board name

kifir23917, 2025-05-12:
From https://github.com/rusefi/rusefi/issues/7829 is my understanding correct, that detailed file names should be used for *.srec file as well as for *.bin file?

rusefillc, 2025-05-12:
@kifir23917 yes, all compiled ECU binaries to stay consistent

rusefillc, 2025-09-07:
Done
for instance rusefi_development_2025-09-07_df8ea2041f48542a382401cdd9ce5f7aef712ae3_update.srec