rusefi/rusefi #7832: TuneCanTool does not produce valid code for GPPWM

rusefillc, 2025-05-04:
open IDEA
invoke TuneCanTool.java main method

FDSoftware, 2025-05-04:
The problem is related to the "getOriginalArrayName" function (`cName` in the debug view), which is returning both the name of the array and the associated parent, using cf.getName() (`cf.name` in the debug view) solves the problem
[image]

FDSoftware, 2025-05-05:
> `gppwm[3].gppwm1_table` should be `gppwm[3].table` no?

After off-github discussion, the correct return would be the following, since the .msq file affects table number 1 and not 4

[image]

now much nicer output of the script (pending to send PR, will edit this to the link)

```c++
static void cannedgppwm2_rpmBins() {
	static const float hardCodedgppwm2_rpmBins[8] = {0.0, 1000.0, 2000.0, 3000.0, 4000.0, 5000.0, 6000.0, 7777.0};
	copyArray(engineConfiguration->gppwm[1].rpmBins, hardCodedgppwm2_rpmBins);
}

static void cannedgppwm2_loadBins() {
	static const float hardCodedgppwm2_loadBins[8] = {0.0, 14.0, 28.0, 40.0, 57.0, 71.0, 85.0, 100.0};
	copyArray(engineConfiguration->gppwm[1].loadBins, hardCodedgppwm2_loadBins);
}

static void cannedgppwm2_table() {
	static const float hardCodedgppwm2_table[8][8] = {
{55,000,	55,000,	55,000,	55,000,	11,000,	11,000,	11,000,	11,000,	},
{55,000,	55,000,	55,000,	55,000,	11,000,	11,000,	11,000,	11,000,	},
{55,000,	55,000,	55,000,	55,000,	11,000,	11,000,	11,000,	11,000,	},
{55,000,	55,000,	55,000,	55,000,	11,000,	11,000,	11,000,	11,000,	},
{55,000,	55,000,	55,000,	55,000,	11,000,	11,000,	11,000,	11,000,	},
{55,000,	55,000,	55,000,	55,000,	11,000,	11,000,	11,000,	11,000,	},
{55,000,	55,000,	55,000,	55,000,	11,000,	11,000,	11,000,	11,000,	},
{55,000,	55,000,	55,000,	55,000,	100,000,	100,000,	100,000,	100,000,	},
};
	copyTable(engineConfiguration->gppwm[1].gppwm2_table, hardCodedgppwm2_table);
}
```

PR #7841 (2025-05-05): improve unit test coverage for TuneCanTool see #7833
resolves #7833 also closes #7832
closed 2025-05-05 by rusefillc