rusefi/rusefi #7840: Burn any changes in TS broken

3eR01ive, 2025-05-05:
**Describe the bug**
Burn any changes in TS broken in last Release for proteus_f7 (20250427) 

**Hardware**
 - Board: [proteus_f7]

**To Reproduce**
Steps to reproduce the behavior:
1. Download and burn release: https://github.com/rusefi/rusefi/releases/tag/release_20250427
2. Open TS
3. Edit any parameter, such as AlternatorVoltageTargetTable
4. Click Burn
5. Get crc check burn fail
6. other parameters can be reset to 0, for example in the screenshot the injector dead time is reset

**Screenshots**
[image]

[image]

[image]

[image]

[image]

[image]

**Additional info**
The penultimate release works correctly! https://github.com/rusefi/rusefi/releases/tag/release_20250203

ggurov, 2025-05-05:
please do a full chip erase and try again

3eR01ive, 2025-05-10:
> please do a full chip erase and try again

of course I did it

3eR01ive, 2025-05-10:
all actions with the new firmware and the old one are the same

rusefillc, 2025-05-16:
@3eR01ive please test fix! appreciate the defect report.