rusefi/rusefi #7850: Suzuki swift 2004-2010 can dash protocol

Tomika012345, 2025-05-06:
I have modified the Mercedes W202 can dash code to work with a 2004-2010 Suzuki Swift(MK2 or RS) dash. 
The suzuki dash uses 5 byte long messages with simple can(non-extended). Can You please implement it as a separate entry? 

This code was compiled and tested on a working car without issues. Some infos in comment.
byte 0 at 0x124 is for warning lights; 

0 - no error / 1 - CEL / 2 - CEL flashing / 3 - nil / 4 - immo / 5 - CEL&immo / 6 - CELflash&immo / 7 - immo / 8 - immo flash / 9 - CEL&immoflash / 10 - CELflash&immoflash

The modified W202 code is below;
---

void canDashboardW202(CanCycle cycle) {
	if (cycle.isInterval(CI::_20ms)) { //factory spec is 10ms, works happily up to 50ms
		{
			CanTxMessage msg(CanCategory::NBC, W202_STAT_1, 5, 0, false); // 0x124 is the suzuki address
			msg[0] = 0x00; // CEL and IMMO warning lights
			msg.setShortValueMsb(Sensor::getOrZero(SensorType::Rpm)*4, 1);
			msg[3] = 0x15; // fixed at that value
			msg[4] = 0xE0; // fixed at that value
		}
	}

	if (cycle.isInterval(CI::_100ms)) { 
		{
			CanTxMessage msg(CanCategory::NBC, W202_STAT_2, 5, 0, false); // 0x310 is the suzuki address
			msg[0] = 0x00; 
			msg[1] = (int)(Sensor::getOrZero(SensorType::VehicleSpeed) / 2);
			msg[2] = 0x00; 
			msg[3] = (int)(Sensor::getOrZero(SensorType::Clt) + 35); // 
			msg[4] = 0x00; 
		}
	}
}

---

rusefillc, 2025-05-06:
I would recommend using https://github.com/rusefi/rusefi/wiki/Lua-Scripting instead

rusefillc, 2025-05-06:
@Tomika012345 yes please attach Lua

F407 versions have low Lua RAM limit, you would have to go F427 or F7

Tomika012345, 2025-05-06:
```
setTickRate(50)
enableCanTx(true)
yourTimer = Timer.new();
yourTimer:reset();

function startup()
surpm = 8000*4

txPayload = {}
txPayload[1] = 0 
txPayload[2] = math.floor(surpm / 256)
txPayload[3] = math.floor(((surpm/256) - math.floor(surpm/256))*256)
txPayload[4] = 21
txPayload[5] = 224
txCan(1, 0x124, 0, txPayload)

txPayload2 = {}
txPayload2[1] = 0 
txPayload2[2] = 100
txPayload2[3] = 0
txPayload2[4] = 0
txPayload2[5] = 0
txCan(1, 0x310, 0, txPayload2)
end

function running()
surpm = getSensor("RPM")*4

  txPayload = {}
txPayload[1] = 0 
  txPayload[2] = math.floor(surpm / 256)
  txPayload[3] = math.floor(((surpm/256) - math.floor(surpm/256))*256)
 txPayload[4] = 21
 txPayload[5] = 224

txCan(1, 0x124, 0, txPayload)

  txPayload2 = {}
txPayload2[1] = 1 
  txPayload2[2] = 0
  txPayload2[3] = 0
 txPayload2[4] = 0
 txPayload2[5] = 0

txCan(1, 0x310, 0, txPayload2)

end

function onTick()

time = yourTimer:getElapsedSeconds()
print(time)

if time < 3 then startup() end
if time > 3 then running() end

end
```

Tomika012345, 2025-05-06:
```
setTickRate(20)
enableCanTx(true)
yourTimer = Timer.new();
yourTimer:reset();
txPayload = {}

function startup()

txPayload[1] = 0 
txPayload[2] = 140
txPayload[3] = 160
txPayload[4] = 21
txPayload[5] = 224
txCan(1, 0x124, 0, txPayload)

txPayload[1] = 0 
txPayload[2] = 100
txPayload[3] = 0
txPayload[4] = 0
txPayload[5] = 0
txCan(1, 0x310, 0, txPayload)

end

function running()
surpm = getSensor("RPM")*4

txPayload[1] = 0 
  txPayload[2] = math.floor(surpm / 256)
  txPayload[3] = math.floor(((surpm/256) - math.floor(surpm/256))*256)
 txPayload[4] = 21
 txPayload[5] = 224

txCan(1, 0x124, 0, txPayload)

txPayload[1] = 0 

if getSensor("VehicleSpeed") == nil then 
  txPayload[2] = 0
else txPayload[2] = getSensor("VehicleSpeed")/2
end

  txPayload[3] = 0
 txPayload[4] =  getSensor("CLT")+35
 txPayload[5] = 0

txCan(1, 0x310, 0, txPayload)

end

function onTick()

if yourTimer:getElapsedSeconds() < 2 then startup() 
else running() end

end
```