rusefi/rusefi #7861: (cranking?) battery problem detection

rusefillc, 2025-05-07:
problem: users are trying to crank with very sad batteries

@ElDominio @stefanst @ggurov would it be legit to throw a FATAL/CRITICAL if battery voltage is between 6 and 7 volts for more than 100ms?

[image]

ggurov, 2025-05-07:
no, lets not disable the ECU in case of low batteries. dying battery is stressful enough, having ECU lock you out of starting the vehicle to make it home to fix it not useful in that case.

what about push starting ?

ElDominio, 2025-05-07:
My PT Cruiser has a similar feature from the factory. There is a battery light on th dash that lights up when similar conditions are detected, but the car is allowed to be driven with the battery on. Critical warnings are very frustrating and usually once the alternator is running, a bad battery can still be used.

Tldr i am not against a warning light (which Harely probably has, need to find battery warning bit over CAN), but I am against ECU death with bad battery

joshImpact25, 2025-05-08:
Harley does have a low battery light. 
 and typically will set a check engine light if voltage drops below 8v while cranking.