rusefi/rusefi #7868: coverage for map averaging integration

rusefillc, 2025-05-10:
let's have coverage for both defects under #7867

fivemeyestoreserver, 2025-05-10:
Your approach to this problem is very elegant. I'm learning a lot from reviewing your code.

FDSoftware, 2025-05-12:
Some tests depend on the number of events scheduled, now after adding the map averaging tests they break since we now have more events, detail here:

```
[==========] 907 tests from 181 test suites ran. (6378 ms total)
[  PASSED  ] 896 tests.
[  FAILED  ] 11 tests, listed below:
[  FAILED  ] misc.testRpmCalculator
[  FAILED  ] trigger.map_cam_by_magic_point
[  FAILED  ] engine.testPlainCrankingWithoutAdvancedFeatures
[  FAILED  ] big.testFuelSchedulerBug299smallAndMedium
[  FAILED  ] big.testFuelSchedulerBug299smallAndLarge
[  FAILED  ] big.testSparkReverseOrderBug319
[  FAILED  ] cranking.testFasterEngineSpinningUp
[  FAILED  ] nissan.vq_vvt
[  FAILED  ] ignition.trailingSpark
[  FAILED  ] OddFireRunningMode.hd
[  FAILED  ] issues.issueOneCylinderSpecialCase968
```
`engine_configuration.cpp` will be updated to the following until the affected tests can be modified
```
#if EFI_PROD_CODE
	engineConfiguration->isMapAveragingEnabled = true;
#endif
```

closed 2025-05-13 by rusefillc