rusefi/rusefi #7870: msg SV#4 fatal updateWidebandFirmware  chEvtGetAndClearEvents

rusefillc, 2025-05-11:
```
Reset Cause: Reset from NRST pin
Power cycle count: 44
Last error type ChibiOS panic
msg SV#4
file ChibiOS/os/rt/src/chsys.c
line 201
SP 0x20020f54
 0x20020f54: 0xfffffa11
 0x20020f50: 0x55555555
 0x20020f4c: 0x0020ad01
 0x20020f48: 0x00000001
 0x20020f44: 0x0020bbd5
 0x20020f40: 0x00000020
 0x20020f3c: 0x00216e6f
 0x20020f38: 0x00000003
 0x20020f34: 0x0027eead
 0x20020f30: 0x2002ed80
```

```
00216e60 <chEvtGetAndClearEvents>:
eventmask_t chEvtGetAndClearEvents(eventmask_t events) {
  216e60:	b510      	push	{r4, lr}
  216e62:	4604      	mov	r4, r0
  216e64:	2320      	movs	r3, #32
  216e66:	f383 8811 	msr	BASEPRI, r3
  _dbg_check_lock();
  216e6a:	f7f4 fea9 	bl	20bbc0 <_dbg_check_lock>
  m = chEvtGetAndClearEventsI(events);
  216e6e:	4620      	mov	r0, r4
  216e70:	f7f9 fa86 	bl	210380 <chEvtGetAndClearEventsI>
  216e74:	4604      	mov	r4, r0
  chSysUnlock();
  216e76:	f7ff ffdb 	bl	216e30 <chSysUnlock.lto_priv.16>
}
```

```
00250250 <_ZN6Engine20periodicSlowCallbackEv>:
void Engine::periodicSlowCallback() {
  250250:	b570      	push	{r4, r5, r6, lr}
  250252:	b082      	sub	sp, #8
  250254:	4605      	mov	r5, r0
	ScopePerf(PE event) : m_event(event) {
  250256:	2006      	movs	r0, #6
  250258:	f88d 0000 	strb.w	r0, [sp]
		perfEventBegin(event);
  25025c:	f041 fac0 	bl	2917e0 <_Z14perfEventBegin2PE>
  if (engineConfiguration->suppressWboWorkaround7048) {
  250260:	4b45      	ldr	r3, [pc, #276]	@ (250378 <_ZN6Engine20periodicSlowCallbackEv+0x128>)
  250262:	f893 3385 	ldrb.w	r3, [r3, #901]	@ 0x385
  250266:	f013 0f01 	tst.w	r3, #1
  25026a:	d00e      	beq.n	25028a <_ZN6Engine20periodicSlowCallbackEv+0x3a>
    static Timer canBusWboSetIndex;
  25026c:	4b43      	ldr	r3, [pc, #268]	@ (25037c <_ZN6Engine20periodicSlowCallbackEv+0x12c>)
  25026e:	681b      	ldr	r3, [r3, #0]
  250270:	f013 0f01 	tst.w	r3, #1
  250274:	d010      	beq.n	250298 <_ZN6Engine20periodicSlowCallbackEv+0x48>
    if (canBusWboSetIndex.getElapsedSeconds() > 1) {
  250276:	4842      	ldr	r0, [pc, #264]	@ (250380 <_ZN6Engine20periodicSlowCallbackEv+0x130>)
  250278:	f016 fe4a 	bl	266f10 <_ZNK5Timer17getElapsedSecondsEv>
  25027c:	eef7 7a00 	vmov.f32	s15, #112	@ 0x3f800000  1.0
  250280:	eeb4 0ae7 	vcmpe.f32	s0, s15
  250284:	eef1 fa10 	vmrs	APSR_nzcv, fpscr
  250288:	dc0d      	bgt.n	2502a6 <_ZN6Engine20periodicSlowCallbackEv+0x56>
	triggerCentral.primaryTriggerConfiguration.update();
  25028a:	f642 20b0 	movw	r0, #10928	@ 0x2ab0
  25028e:	4428      	add	r0, r5
  250290:	f005 fe66 	bl	255f60 <_ZN20TriggerConfiguration6updateEv>
	for (int camIndex = 0;camIndex < CAMS_PER_BANK;camIndex++) {
  250294:	2400      	movs	r4, #0
  250296:	e017      	b.n	2502c8 <_ZN6Engine20periodicSlowCallbackEv+0x78>
    static Timer canBusWboSetIndex;
  250298:	4839      	ldr	r0, [pc, #228]	@ (250380 <_ZN6Engine20periodicSlowCallbackEv+0x130>)
  25029a:	f016 fda9 	bl	266df0 <_ZN5TimerC1Ev>
  25029e:	4b37      	ldr	r3, [pc, #220]	@ (25037c <_ZN6Engine20periodicSlowCallbackEv+0x12c>)
  2502a0:	2201      	movs	r2, #1
  2502a2:	601a      	str	r2, [r3, #0]
  2502a4:	e7e7      	b.n	250276 <_ZN6Engine20periodicSlowCallbackEv+0x26>
      canBusWboSetIndex.reset();
  2502a6:	4836      	ldr	r0, [pc, #216]	@ (250380 <_ZN6Engine20periodicSlowCallbackEv+0x130>)
  2502a8:	f014 fe0a 	bl	264ec0 <_ZN5Timer5resetEv>
      setWidebandOffset(3);
  2502ac:	2003      	movs	r0, #3
  2502ae:	f02e fdf7 	bl	27eea0 <_Z17setWidebandOffseth>
```

dron0gus, 2025-05-12:
I'm not sure if we need to keep this workaround, or we just need to fix WBO firmware?

rusefillc, 2025-05-12:
this is about @ElDominio hitting a rare option to disable broadcast which is rarely used, root cause is lack of WBO can bus ID control