rusefi/rusefi #7875: ADC stability broken by EFI_INTERNAL_SLOW_ADC_BACKGROUND

ggurov, 2025-05-12:
**Describe the bug**
with EFI_INTERNAL_SLOW_ADC_BACKGROUND TRUE

[image]

without it

[image]

**Hardware**
 - Board: F7 lqfp208 board
 - Engine info NA
 - 
**To Reproduce**
Steps to reproduce the behavior:
1. merge in changes, see random dropouts in adc readings
2. revert
3. signal is stable

closed 2025-05-12 by ggurov

rusefillc, 2025-05-12:
@ggurov please clarify why closed?

ggurov, 2025-05-12:
i saw part of the commit i'm not sure i applied. will verify and reopen uf still broken

ggurov, 2025-05-12:
👍

dron0gus, 2025-05-12:
@ggurov please pick fresh ChibiOS. It has fixed ADC driver.

ggurov, 2025-05-12:
@dron0gus fresher than what master's .gitmodules calls for ?

dron0gus, 2025-05-12:
@ggurov not sure about step 1 in first post of this issue... So make sure you have https://github.com/rusefi/ChibiOS/commit/35b1d17163d272cb76bff58e28970c2d02bc2c06 in your ChibiOS.

ggurov, 2025-05-12:
@dron0gus thank you, 

i've added 

```git submodule update --remote```

to my workflow to prevent this in the future